triSYCL/sycl #10: [SYCL] POCL SPIR Builtin Translation

agozillon, 2019-04-15:
As an extension to https://github.com/triSYCL/sycl/issues/9 there are currently some incorrectly mangled SPIR builtins in POCL (https://github.com/pocl/pocl/issues/698) that will need to be handled for clean execution of SYCL spir-df output by a POCL runtime. 

The incorrect mangling appears to be caused by some type aliasing via using declarations in POCL's _builtin_renames.h file and currently affects a significant amount of math functions, some atomics and printf. 

I can currently see thing being handled in one of two ways:
- Rework the existing POCL implementation to use the correct function names and then put in a pull request to try and change the upstream implementation. This would be the ideal long-term avenue, but may be very difficult as there will most likely be some reason behind the renaming of these builtins that could be a far reaching problem.
- Alter the InSPIRation pass to optionally translate the correct SPIR mangled names to the altered POCL renamed builtins, effectively by passing the need to alter POCL. The easier of the two options but inflicts a certain amount of code debt on us and if the POCL issue gets fixed then we'll have to alter the InSPIRation pass again.

franz, 2019-09-06:
Hello,

This was _partially_ resolved in git master. There is now a script, which generates a "wrapper" with correctly mangled SPIR names, calling the pocl's kernel library functions. Partially resolved because it does not contain every possible SPIR opencl name (vectorized ones are missing, plus a few others), and it's only for the CPU device.

> some reason behind the renaming of these builtins that could be a far reaching problem

The original reason (at least IIRC) was that many kernel library functions were calling libc's math functions, so to avoid name clashes, pocl renamed kernel functions with macros. This is no longer an issue as pocl doesn't use libm anymore. But unfortunately removing the rename macros doesn't solve the problem. There are two reasons:

1. Argument address spaces. SPIR has fixed address space numbers, but pocl's kernel library uses *target* address spaces, which are different for every device. E.g. in the library for CPU device, all OpenCL AS map to a single target AS: zero; while library for CUDA uses SPIR numbers (IIRC). This has to be handled somewhere by address casts.

2. The calling convention. SPIR has its own LLVM calling convention (spir_func), while pocl's kernel library use their target's. Certain passes in LLVM have a feature, where if a function is called with the wrong convention, it's replaced by "llvm_unreachable".

So for these reasons, i've created a python script which generates said wrapper. Please give it a try (via pocl git master) when you have time.

agozillon, 2019-09-09:
Thank you very much for looking into this and updating us, I'll take a look into this to make sure when I get a chance! And thanks for the detailed breakdown of the problem.