triSYCL/sycl #17: [SYCL] Experiment with Compile Time Regular Expressions in SYCL

agozillon, 2019-04-15:
Perhaps an interesting little blue skies research project for C++20. We rely a lot on regex's at the moment, so a very interesting C++ project for the runtime properties (and perhaps the compiler in some way) may be: https://github.com/hanickadot/compile-time-regular-expressions by Hana Dusíková

As we currently regex at runtime over the constexpr mangled kernel name to rip properties from it, doing this at compile time may offset a lot of the run-time impact that we incur.

MathiasMagnus, 2019-04-20:
I totally agree. Anything we can do at compile-time, we should. GCC-Clang-MSVC all have fairly good support for CTRE.

One thing to consider is: We are currently fairly keen on bumping the required language standard version, but should triSYCL settle for something like C++17 as a minimum, all dependencies must compile on that standard as well, and CTRE is pretty modern. I don't know about their policy regarding language versions.

agozillon, 2019-04-22:
That's true, pushing the minimum to C++2a may not be ideal at the moment, I do think Ronan would like to use some C++2a related things though in the near future and it may be a little bit painful to maintain the runtime long term with a lot of optional defines. @keryell what do you think about having the minimum as C++17?

As for CTRE yes there is a lot of support and still more to come in the future (hopefully), I believe the CTRE library by Hana Dusíková does have the ability to be used with C++17 with some slightly worse syntax and I don't think the C++20 variation of the syntax is supported in Clang just yet as it relies on some WIP C++20 features (Class types in non-type template parameters). There is a proposal for the library but it looks to be C++23 (but of course, not guaranteed in anyway as it's r0... many more hurdles):  http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2019/p1433r0.pdf  

So at least in this case, it would be more testing the usage of the CTRE library or the concepts within over the boost library's regex facilities that we currently use. It's just a small idea I had that interested me and might be interesting to look into if we use more regex's in the long term. Perhaps, it's not a good one though, especially if we plan to maintain a C++17 minimum standard.

keryell, 2019-05-22:
Give me some numbers about the fact that CTRE will bring any thing noticeable in our context... :-)
Otherwise, I need C++2a only on some part of triSYCL which are not public yet.