triSYCL/sycl #21: [SYCL] Merge KernelNameGen pass and InSPIRation pass

agozillon, 2019-04-18:
KernelNameGen is a fairly simple LLVM pass at the moment that generates a list of kernel names to be used by the sycl-xocc shell script, it should be possible to refactor the functionality from here into InSPIRation to make maintenance simpler in the long term.

keryell, 2019-04-19:
I do not know if it is very interesting to have everything in the same pass.
Up to now I was thinking that having a lot off small passes would help to build new compilers for more targets, just by chaining the passes.

agozillon, 2019-04-20:
That makes sense, although I think InSPIRation currently does some things that wouldn't be required for a target that consumes SPIR for example something like POCL could make use of the conversion from SPIRV to SPIR builtins. But it wouldn't have need for the conversion of the kernel names to a hash like xocc currently requires.

So perhaps the inverse would be better, where existing functionality from InSPIRation is broken into something more fine grain to allow for better composition of passes in the long term?

agozillon, 2019-05-16:
No longer relevant with the future merge of: https://github.com/triSYCL/sycl/pull/27

KernelNameGen has become a `property` generator for kernels, gathering useful kernel information to feed to xocc via script.

closed 2019-05-16 by agozillon