triSYCL/sycl #28: Compiler crash with named function object

j-stephan, 2019-05-20:
I am trying to compile the attached code (simple N-body) with the current versions of the compiler (up-to-date `sycl/unified/master`) and SDAccel (2018.3) and a self-compiled XRT (up-to-date `2018.3`), OS is Ubuntu 18.10. Upon starting the compilation it crashed almost immediately with the following message:

```
$ clang++ -std=c++2a -fsycl -fsycl-targets=fpga64-xilinx-unknown-sycldevice nbody.cpp -o nbody.sw_emu -lOpenCL
Stack dump:
0.	Program arguments: /home/jan/software/sycl/bin/opt -O3 -asfix -globaldce -inSPIRation -globaldce -kernelNameGen /tmp/nbody-4bd090.o -o /tmp/nbody_kernels-optimized.bc 
1.	Running pass 'ASFixer' on module '/tmp/nbody-4bd090.o'.
 #0 0x0000562b936c72ca llvm::sys::PrintStackTrace(llvm::raw_ostream&) (/home/jan/software/sycl/bin/opt+0x20932ca)
 #1 0x0000562b936c5194 llvm::sys::RunSignalHandlers() (/home/jan/software/sycl/bin/opt+0x2091194)
 #2 0x0000562b936c5315 SignalHandler(int) (/home/jan/software/sycl/bin/opt+0x2091315)
 #3 0x00007f7f8a12fdd0 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x12dd0)
 #4 0x0000562b930e1fa3 llvm::Value::getContext() const (/home/jan/software/sycl/bin/opt+0x1aadfa3)
 #5 0x0000562b930e395a llvm::ValueHandleBase::AddToUseList() (/home/jan/software/sycl/bin/opt+0x1aaf95a)
 #6 0x0000562b93701113 llvm::ValueMap<llvm::Value const*, llvm::WeakTrackingVH, llvm::ValueMapConfig<llvm::Value const*, llvm::sys::SmartMutex<false> > >::operator[](llvm::Value const* const&) (.isra.418) (/home/jan/software/sycl/bin/opt+0x20cd113)
 #7 0x0000562b937048bc llvm::CloneFunctionInto(llvm::Function*, llvm::Function const*, llvm::ValueMap<llvm::Value const*, llvm::WeakTrackingVH, llvm::ValueMapConfig<llvm::Value const*, llvm::sys::SmartMutex<false> > >&, bool, llvm::SmallVectorImpl<llvm::ReturnInst*>&, char const*, llvm::ClonedCodeInfo*, llvm::ValueMapTypeRemapper*, llvm::ValueMaterializer*) (/home/jan/software/sycl/bin/opt+0x20d08bc)
 #8 0x0000562b9391d1fb (anonymous namespace)::createNewFunction(llvm::Function*, llvm::FunctionType*, llvm::DenseMap<llvm::Value*, llvm::Value*, llvm::DenseMapInfo<llvm::Value*>, llvm::detail::DenseMapPair<llvm::Value*, llvm::Value*> >&, llvm::DenseMap<llvm::User*, llvm::SmallVector<std::pair<unsigned int, llvm::Value*>, 32u>, llvm::DenseMapInfo<llvm::User*>, llvm::detail::DenseMapPair<llvm::User*, llvm::SmallVector<std::pair<unsigned int, llvm::Value*>, 32u> > >&) (/home/jan/software/sycl/bin/opt+0x22e91fb)
 #9 0x0000562b93920637 (anonymous namespace)::doReplace(llvm::DenseMap<llvm::Value*, llvm::Value*, llvm::DenseMapInfo<llvm::Value*>, llvm::detail::DenseMapPair<llvm::Value*, llvm::Value*> >&, llvm::DenseMap<llvm::User*, llvm::SmallVector<std::pair<unsigned int, llvm::Value*>, 32u>, llvm::DenseMapInfo<llvm::User*>, llvm::detail::DenseMapPair<llvm::User*, llvm::SmallVector<std::pair<unsigned int, llvm::Value*>, 32u> > >&, llvm::DenseMap<llvm::Function*, llvm::FunctionType*, llvm::DenseMapInfo<llvm::Function*>, llvm::detail::DenseMapPair<llvm::Function*, llvm::FunctionType*> >&) (/home/jan/software/sycl/bin/opt+0x22ec637)
#10 0x0000562b93923ed1 (anonymous namespace)::ASFixer::runOnModule(llvm::Module&) (/home/jan/software/sycl/bin/opt+0x22efed1)
#11 0x0000562b9308fa02 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/jan/software/sycl/bin/opt+0x1a5ba02)
#12 0x0000562b91c4e70f main (/home/jan/software/sycl/bin/opt+0x61a70f)
#13 0x00007f7f897d509b __libc_start_main /build/glibc-B9XfQf/glibc-2.28/csu/../csu/libc-start.c:308:16
#14 0x0000562b91cdc51a _start (/home/jan/software/sycl/bin/opt+0x6a851a)
/home/jan/software/sycl/bin/sycl-xocc: line 76:  9118 Segmentation fault      (core dumped) $OPT -O3 -asfix -globaldce -inSPIRation -globaldce -kernelNameGen "$4" -o "$5/$3_kernels-optimized.bc"
/home/jan/software/sycl/bin/llvm-link: /tmp/nbody_kernels-optimized.bc: error: Could not open input file: No such file or directory
/home/jan/software/sycl/bin/llvm-link: error:  loading file '/tmp/nbody_kernels-optimized.bc'
/home/jan/software/sycl/bin/sycl-xocc: line 87: /tmp/KernelNames_nbody.txt: No such file or directory
ERROR: No input file specified.
Allowed options:
[...]
/usr/bin/ld: /tmp/nbody-d900a3.o: file not recognized: file truncated
clang-9: error: sycl-link-xocc command failed with exit code 1 (use -v to see invocation)
clang-9: error: linker command failed with exit code 1 (use -v to see invocation)
```

I suppose something is still wrong with my code but I assume that a compiler crash isn't the intended behaviour here ;-)
[nbody.zip](https://github.com/triSYCL/sycl/files/3198364/nbody.zip)

j-stephan, 2019-05-20:
Oh, by the way (don't want to open a new issue for this): Ubuntu 18.10's support will end in a few weeks. Can I safely upgrade to 19.04 and still use the SYCL toolchain?

keryell, 2019-05-20:
Nice to see that the compiler crashed in `ASFixer` and not before...  :-)
I never use the functor kernels and perhaps nobody tried with a default constructor.
We will look at this.
Have you tried your code with the default Intel compiler first, by the way? The problem might be upstream.

For the Ubuntu 18.10 story, we are working on it. Just waiting for a colleague to provide me a new machine with a new FPGA card to ease the transition... Thanks.

agozillon, 2019-05-21:
Thanks for the report, this https://github.com/triSYCL/sycl/pull/29 should fix it, but it's a fair bit passed the end of the work day here so probably won't be merged till tomorrow. It is just a one line fix though inside the script so shouldn't be difficult to self apply (sorry for the trouble).

Just swaps the ASFixer and O3 pass around which allows your test to compile in sw emu on my machine! I believe this is because the ASFixer pass was never intended to be ran after O3 optimizations as Intels implementation targets SPIRV which doesn't use the usual opt pass pipeline (ASFixer comes from the upstream implementation, rather than us in this case). As an aside, if you want to debug things to do with xocc or add new xocc specific passes this script is the place to do it.

Unfortunately there appears to be an error in your example at the moment so it doesn't execute correctly, a runtime assertion is hit for both Intel and Xilinx compilation (it also deadlocks in the Upstream SYCL master branch which has a new scheduler). I'd imagine it's something to do with incorrect usage of the accessors rather than a bug with the SYCL implementation though (but if it is a bug feel free to open up an issue, although it may be better suited to the Intel Upstream implementation in this case), I was tempted to try fix it but it's a bit late in the day sadly! The SYCL spec references here: https://github.com/intel/llvm/issues/143 might be useful as I made a mistake myself recently with accessors/buffers and incorrectly opened an issue!

We'd be interested (and thankful) in adding that test to our current set of tests if you'd be interested in doing so, once you've completed it (just some minor alterations and a pull request should be enough!)? As it's quite a nice test, complex and concise! If not then that's no problem though.

agozillon, 2019-05-21:
Issue should be fixed with merge of: https://github.com/triSYCL/sycl/pull/29 if I am mistaken and it's not feel free to reopen the issue!

closed 2019-05-21 by agozillon

j-stephan, 2019-05-23:
Sorry for the delay, other work got in the way the past days. I can confirm that the compiler no longer crashes.

> We'd be interested (and thankful) in adding that test to our current set of tests if you'd be interested in doing so, once you've completed it (just some minor alterations and a pull request should be enough!)? As it's quite a nice test, complex and concise! If not then that's no problem though.

I'll be happy to contribute once everything works. I originally intended to use this code for playing around with the Xilinx extensions (pipelining). Would you be interested in the (present) vanilla version or the one with extensions?

agozillon, 2019-05-23:
The extensions variant sounds like it would be most interesting, but whenever you feel the example is complete to your satisfaction is great.

keryell, 2019-06-06:
@j-stephan We have a new machine with Ubuntu 19.04 and a Xilinx Alveo U200 board now.
There is a recipe coming, if you are interested: https://github.com/triSYCL/sycl/pull/33

j-stephan, 2019-06-06:
Great, thank you!