triSYCL/sycl #31: Missing access mode: discard_write

j-stephan, 2019-05-23:
Using the same N-body code as in #28 I receive the following failed assertion once the control flow reaches the first host-to-device copy:

```
nbody.sw_emu: /home/jstephan/software/sycl/lib/clang/9.0.0/include/CL/sycl/detail/buffer_impl.hpp:519: size_t cl::sycl::detail::buffer_impl<std::allocator<char> >::convertSycl2OCLMode(cl::sycl::access::mode) [AllocatorT = std::allocator<char>]: Assertion `0 && "Unhandled conversion from Sycl access mode to OCL one."' failed.
```

And the mode is indeed missing from `buffer_impl.hpp`. Is this a feature waiting for implementation? I haven't found anything related in the issues here or the original triSYCL repository.

agozillon, 2019-05-23:
Once this is merged into Unified/Next https://github.com/triSYCL/sycl/pull/30 (should be end of Today or tomorrow PDT), I believe discard_write should exist and Unified/Next should be somewhat 'stable', I'll double check though and add an edit note if i am incorrect. The master branch is a fair bit behind the main Intel SYCL branch at the moment unfortunately.

agozillon, 2019-06-03:
Going to close this as this should now exist in the unified-next branch (if it doesn't or you have some other problems with it, feel free to reopen this issue or create a new one).

However, hw_emu is still a little ropy on that branch unfortunately at least with the new SDAccel 2019.1 release, we're working on it, but if you encounter any new bugs please do open an issue.

closed 2019-06-03 by agozillon

j-stephan, 2019-06-04:
Okay, I will upgrade to `sycl/unified/next` and 2019.1. Is the next-branch the development branch or will it be merged into master in the future?

agozillon, 2019-06-04:
It's the development branch for just now, but it will eventually be merged into sycl/unified/master or replace it.