triSYCL/sycl #57: std::make_tuple inside a lambda does not compile

Victorlzd, 2019-07-24:
Problem when compiling a call to `std::make_tuple` inside a lambda in a kernel.

The following code don't compile : 
```
cgh.single_task<class array_add>([=]() {
      auto f = [&] (auto i) {buff_w[0] = std::get<0>(std::make_tuple(0));};
      f(0);
 });
```
and `xocc` emits the following error message : 
```
ERROR: [XOCC 200-41] in function 'xSYCL12199936647150665982': unsupported pointer reinterpretation from type 'i64*' to type 'i32*'.
ERROR: [XOCC 200-71] in function 'xSYCL12199936647150665982': function 'FORWARD_REFERENCE&& std::forward<int>(std::remove_reference<FORWARD_REFERENCE>::type&)' has no function body.
ERROR: [XOCC 200-71] in function 'xSYCL12199936647150665982': function 'std::tuple_element<FORWARD_REFERENCE, std::tuple<FORWARD_REFERENCE' has no function body.
ERROR: [XOCC 200-71] in function 'xSYCL12199936647150665982': function 'FORWARD_REFERENCE&& std::forward<int&&>(std::remove_reference<FORWARD_REFERENCE>::type&)' has no function body.
ERROR: [XOCC 200-70] Synthesizability check failed.
ERROR: [XOCC 60-300] Failed to build kernel(ip) xSYCL12199936647150665982, see log for details: /var/tmp/victorl/result/issue_tuple_lambda/hw/2019-07-24_14_10/_x/xSYCL12199936647150665982/xSYCL12199936647150665982/vivado_hls.log
ERROR: [XOCC 60-599] Kernel compilation failed to complete
ERROR: [XOCC 60-592] Failed to finish compilation
```
The following code is compiling and working without any issue : 
```
cgh.single_task<class array_add>([=]() {
      auto f = [&] () {buff_w[0] = std::get<0>(std::make_tuple(0));};
      f();
});
```

Ralender, 2020-12-02:
This seems to be compiling and executing fine.

closed 2020-12-02 by Ralender

keryell, 2020-12-03:
Cool.