triSYCL/sycl #61: [SYCL] fpga_pipes.cpp pipe ambiguity error with unistd.h pipe

agozillon, 2019-07-27:
Seems to be something incorrect in the Clang/LLVM/libsycl.so build process or some incorrect system inclusion as part of the SYCL compilation process causing unistd.h to be added (could just be an out of order inclusion). This isn't a problem when using an unmodified Intel SYCL compiler (although technically were behind a few patches so maybe a follow up patch I don't recall fixed it). 

It results in the following:

/storage/ogozillo/intel-sycl/sycl/build/lib/clang/9.0.0/include/CL/sycl/pipes.hpp:18:68: note: candidate found by name lookup is 'cl::sycl::pipe'
template <class name, class dataT, int32_t min_capacity = 0> class pipe {
                                                                   ^
fpga_pipes.cpp:141:11: error: reference to 'pipe' is ambiguous
          pipe<class pipe_type_for_lambdas, int>::read(SuccessCode);
          ^
/usr/include/unistd.h:417:12: note: candidate found by name lookup is 'pipe'
extern int pipe (int __pipedes[2]) __THROW __wur;
           ^

Workaround is just to always be specific that you're using SYCL's pipe i.e. longhand cl::sycl::pipe.

keryell, 2019-09-19:
I guess we can close this when we merge with the upstream version, then.

agozillon, 2019-09-19:
should be able to yes

closed 2019-09-19 by agozillon