triSYCL/sycl #62: [SYCL] Compiler crashes with no input files

j-stephan, 2019-07-31:
This is somewhat obscure. The command line

```
clang++ -fsycl foo.cpp -lbla
```

will produce a segmentation fault if the compiler can't find `foo.cpp`. This only occurs when trying to link a library, the command line without `-lbla` will just tell me that there is no `foo.cpp`.

Error message:

```
clang-9: error: no such file or directory: 'foo.cpp'                       
Stack dump:                                                                                                                                                                                                
0.      Program arguments: clang++ -fsycl foo.cpp -lbla
1.      Compilation construction
2.      Building compilation jobs                                                                                                                                                                          
3.      Building compilation jobs                                                                 
4.      Building compilation jobs
5.      Building compilation jobs
6.      Computing output path                   
 #0 0x00005610f535c83a llvm::sys::PrintStackTrace(llvm::raw_ostream&) (/home/jan/software/sycl/xilinx/bin/clang-9+0x257e83a)                                                                               
 #1 0x00005610f535a554 llvm::sys::RunSignalHandlers() (/home/jan/software/sycl/xilinx/bin/clang-9+0x257c554)
 #2 0x00005610f535a6d5 SignalHandler(int) (/home/jan/software/sycl/xilinx/bin/clang-9+0x257c6d5)
 #3 0x00007fb6ace00f40 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x13f40)
 #4 0x00007fb6ac9ec0c7 /build/glibc-KRRWSm/glibc-2.29/string/../sysdeps/x86_64/multiarch/strlen-avx2.S:96:0
 #5 0x00005610f58c5c45 clang::driver::Driver::GetNamedOutputPath(clang::driver::Compilation&, clang::driver::JobAction const&, char const*, llvm::StringRef, bool, bool, llvm::StringRef) const (/home/jan/s
oftware/sycl/xilinx/bin/clang-9+0x2ae7c45)
 #6 0x00005610f58c9a50 clang::driver::Driver::BuildJobsForActionNoCache(clang::driver::Compilation&, clang::driver::Action const*, clang::driver::ToolChain const*, llvm::StringRef, bool, bool, char const*
, std::map<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, clang::driver::InputInfo, std::less<std::pair<clang::driver::Action co
nst*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::allocator<std::pair<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_trait
s<char>, std::allocator<char> > > const, clang::driver::InputInfo> > >&, clang::driver::Action::OffloadKind) const (/home/jan/software/sycl/xilinx/bin/clang-9+0x2aeba50)
 #7 0x00005610f58ca8f7 clang::driver::Driver::BuildJobsForAction(clang::driver::Compilation&, clang::driver::Action const*, clang::driver::ToolChain const*, llvm::StringRef, bool, bool, char const*, std::
map<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, clang::driver::InputInfo, std::less<std::pair<clang::driver::Action const*, s
td::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::allocator<std::pair<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>
, std::allocator<char> > > const, clang::driver::InputInfo> > >&, clang::driver::Action::OffloadKind) const (/home/jan/software/sycl/xilinx/bin/clang-9+0x2aec8f7)
 #8 0x00005610f58c8a56 clang::driver::Driver::BuildJobsForActionNoCache(clang::driver::Compilation&, clang::driver::Action const*, clang::driver::ToolChain const*, llvm::StringRef, bool, bool, char const*
, std::map<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, clang::driver::InputInfo, std::less<std::pair<clang::driver::Action co
nst*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::allocator<std::pair<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_trait
s<char>, std::allocator<char> > > const, clang::driver::InputInfo> > >&, clang::driver::Action::OffloadKind) const (/home/jan/software/sycl/xilinx/bin/clang-9+0x2aeaa56)
 #9 0x00005610f58ca8f7 clang::driver::Driver::BuildJobsForAction(clang::driver::Compilation&, clang::driver::Action const*, clang::driver::ToolChain const*, llvm::StringRef, bool, bool, char const*, std::
map<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, clang::driver::InputInfo, std::less<std::pair<clang::driver::Action const*, s
td::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::allocator<std::pair<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>
, std::allocator<char> > > const, clang::driver::InputInfo> > >&, clang::driver::Action::OffloadKind) const (/home/jan/software/sycl/xilinx/bin/clang-9+0x2aec8f7)
#10 0x00005610f58ccd34 void llvm::function_ref<void (clang::driver::Action*, clang::driver::ToolChain const*, char const*)>::callback_fn<clang::driver::Driver::BuildJobsForActionNoCache(clang::driver::Com
pilation&, clang::driver::Action const*, clang::driver::ToolChain const*, llvm::StringRef, bool, bool, char const*, std::map<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::c
har_traits<char>, std::allocator<char> > >, clang::driver::InputInfo, std::less<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >,
 std::allocator<std::pair<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const, clang::driver::InputInfo> > >&, clang::driver::Ac
tion::OffloadKind) const::'lambda0'(clang::driver::Action*, clang::driver::ToolChain const*, char const*)>(long, clang::driver::Action*, clang::driver::ToolChain const*, char const*) (/home/jan/software/s
ycl/xilinx/bin/clang-9+0x2aeed34)
#11 0x00005610f59c7c54 clang::driver::OffloadAction::doOnEachDeviceDependence(llvm::function_ref<void (clang::driver::Action*, clang::driver::ToolChain const*, char const*)> const&) const (/home/jan/softw
are/sycl/xilinx/bin/clang-9+0x2be9c54)
#12 0x00005610f58c824a clang::driver::Driver::BuildJobsForActionNoCache(clang::driver::Compilation&, clang::driver::Action const*, clang::driver::ToolChain const*, llvm::StringRef, bool, bool, char const*
, std::map<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, clang::driver::InputInfo, std::less<std::pair<clang::driver::Action co
nst*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::allocator<std::pair<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_trait
s<char>, std::allocator<char> > > const, clang::driver::InputInfo> > >&, clang::driver::Action::OffloadKind) const (/home/jan/software/sycl/xilinx/bin/clang-9+0x2aea24a)
#13 0x00005610f58ca8f7 clang::driver::Driver::BuildJobsForAction(clang::driver::Compilation&, clang::driver::Action const*, clang::driver::ToolChain const*, llvm::StringRef, bool, bool, char const*, std::
map<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, clang::driver::InputInfo, std::less<std::pair<clang::driver::Action const*, s
td::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::allocator<std::pair<std::pair<clang::driver::Action const*, std::__cxx11::basic_string<char, std::char_traits<char>
, std::allocator<char> > > const, clang::driver::InputInfo> > >&, clang::driver::Action::OffloadKind) const (/home/jan/software/sycl/xilinx/bin/clang-9+0x2aec8f7)
#14 0x00005610f58cae50 clang::driver::Driver::BuildJobs(clang::driver::Compilation&) const (/home/jan/software/sycl/xilinx/bin/clang-9+0x2aece50)
#15 0x00005610f58cc6a5 clang::driver::Driver::BuildCompilation(llvm::ArrayRef<char const*>) (/home/jan/software/sycl/xilinx/bin/clang-9+0x2aee6a5)
#16 0x00005610f391f467 main (/home/jan/software/sycl/xilinx/bin/clang-9+0xb41467)
#17 0x00007fb6ac88cb6b __libc_start_main /build/glibc-KRRWSm/glibc-2.29/csu/../csu/libc-start.c:342:3
#18 0x00005610f39a5aca _start (/home/jan/software/sycl/xilinx/bin/clang-9+0xbc7aca)
[1]    25391 segmentation fault (core dumped)  clang++ -fsycl foo.cpp -lbla
```

I'm on the latest commit of the `master` branch (commit c3c6023f).

I don't have the Intel compiler available right now, so I can't confirm if this is specific to this compiler or is an upstream issue.

agozillon, 2019-07-31:
Yeah. It's a known problem, I believe it was fixed in the upstream Intel compiler (but I'd double check just in case as it would be useful for them to know if it's a problem as well), but persists for us at the moment. I didn't realize it was linked to the addition of libraries though which is helpful information! 

Thank you for opening the issue!

j-stephan, 2020-11-27:
This is no longer an issue.

closed 2020-11-27 by j-stephan

keryell, 2020-12-01:
Nice to see problems solved! \o/