triSYCL/sycl #64: [SYCL] Compiler forgets typedefs

j-stephan, 2019-07-31:
In `sw_emu` mode (didn't test the others) `clang` will forget the integer typedefs inside `namespace std` after `xocc` has completed its tasks.

Example code:

```c++
#include <cstdint>
#include <CL/sycl.hpp>

template <std::uint32_t Var>
struct foo
{
    cl::sycl::accessor<int, 1, cl::sycl::access::mode::read_write> acc;

    auto operator()()
    {
        for(auto i = 0; i < 1024; ++i)
            acc[i] = Var;
    }
};

struct xilinx_selector : public cl::sycl::device_selector
{
    auto operator()(const cl::sycl::device& dev) const -> int
    {
        const auto vendor = dev.get_info<cl::sycl::info::device::vendor>();
        return (vendor.find("Xilinx") != std::string::npos) ? 1 : -1;
    }
};

auto main() -> int
{
    auto queue = cl::sycl::queue{xilinx_selector{}};

    auto buf = cl::sycl::buffer<int>{cl::sycl::range<1>{1024}};

    queue.submit([&](cl::sycl::handler& cgh)
    {
        auto acc = buf.get_access<cl::sycl::access::mode::read_write>(cgh);

        auto kernel = foo<42>{acc};
        cgh.single_task<foo<42>>(kernel);
    });
    queue.wait();
}
```

Command line:

```
clang++ -std=c++2a -fsycl -fsycl-targets=fpga64-xilinx-unknown-sycldevice  test.cpp -lOpenCL
```

Output:

```
[xocc output snipped]
In file included from <built-in>:1:
/tmp/test-9a9f3e.h:6:16: error: no type named 'uint32_t' in namespace 'std'
template <std::uint32_t Var> struct foo;
          ~~~~~^
test.cpp:5:25: error: template non-type parameter has a different type 'std::uint32_t' (aka 'unsigned int') in template redeclaration
template <std::uint32_t Var>
                        ^
/tmp/test-9a9f3e.h:6:25: note: previous non-type template parameter with type 'int' is here
template <std::uint32_t Var> struct foo;
                        ^
test.cpp:36:23: error: implicit instantiation of undefined template 'foo<42>'
        auto kernel = foo<42>{acc};
                      ^
/tmp/test-9a9f3e.h:6:37: note: template is declared here
template <std::uint32_t Var> struct foo;
                                    ^
3 errors generated.
```

keryell, 2019-08-01:
It might be a bug in the Intel SYCL compiler when they handle the integration header.
There were some fixes on their side that have not landed in our version yet...
So it might have been fixed. This should be merged in soon.

agozillon, 2019-08-01:
I looked into it a bit, it breaks on my up-to-date Intel SYCL version, so I was incorrect and it appears the fix I thought applied here doesn't. So this might be one for the folks upstream to have a look into, @bader sorry to ping you, but this issue may be if interest to you.

j-stephan, 2019-08-01:
This seems to extend to compiler-defined types as well. `clang` is complaining about some `_Bool` type used as a non-type template parameter, but `_Bool` doesn't appear anywhere in the code base (`bool` does, though). I'm still working on a minimal example, simply replacing the `uint32_t` above doesn't do the trick.

j-stephan, 2019-08-07:
Actually the `_Bool` type problem doesn't seem to be related to this issue as I found out today. See this issue at Intel's repo for further reference: intel/llvm#469

alexbatashev, 2019-08-07:
@j-stephan are you sure nested templates are the case here?

```cpp
#include <CL/sycl.hpp>

int main() {
    cl::sycl::queue q;
    q.submit([=](cl::sycl::handler &cgh) {
              cgh.single_task<bool>([=] () {});
            });
    return 0;
}
```

fails with the same error. Can you please check if nested templates work for you with int/float?

j-stephan, 2019-08-07:
@alexbatashev Apparently `_Bool` seems to be something different altogether, looks like you got yourself a minimal example there :-) Nested templates still fail with other types, though.

keryell, 2019-08-08:
I do not remember whether non-user defined data types were ever thought of in the specification...
It should probably work if we do not try to forward declare them... :-)
In the glossary of the current specification:
> kernel name: A kernel name is a class type that is used to assign a name to the kernel function, used to link the host system with the kernel object output by the device compiler.
I am unsure that "class object" means anything in the ISO C++ jargon and it looks more like an overlook when the specification glossary was collected.

For orthogonality, I think we should allow any type, as suggested by the specification, except in the glossary.

agozillon, 2019-10-08:
I think this is now fixed and close able in the sycl/unified/next branch. Feel free to give it a try when you get a spare minute and close this issue and/or close the issue and open it again if you find a problem!

j-stephan, 2019-10-10:
Yup, working for `sw_emu`. Closing.

j-stephan, 2019-10-10:
Reopening for now, since this is also causing an ICE for `hw_emu`. Please close if this entirely unrelated.

```
[INFO] Could not find v++ executable in /opt/xilinx/SDx/2019.1/bin
[INFO] Try with xocc...
warning: Linking two modules of different data layouts: '/opt/xilinx/SDx/2019.1/bin/../lnx64/lib/libspir64-39-hls.bc' is 'e-m:e-i64:64-i128:128-i256:256-i512:512-i1024:1024-i2048:2048-i4096:4096-n8:16:32:64-S128-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024' whereas 'llvm-link' is 'e-i64:64-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024'

warning: Linking two modules of different target triples: /opt/xilinx/SDx/2019.1/bin/../lnx64/lib/libspir64-39-hls.bc' is 'fpga64-xilinx-none' whereas 'llvm-link' is 'spir64'

Invoking Kernel Compilation with /opt/xilinx/SDx/2019.1/bin/xocc
--target: hw_emu
--platform: xilinx_u200_xdma_201830_2
Compiling kernel: xSYCL16676224068994425510
Outputting file to: /tmp/xSYCL16676224068994425510.xo
Input file is: /tmp/main_kernels-linked.xpirbc
Option Map File Used: '/opt/xilinx/SDx/2019.1/data/sdx/xocc/optMap.xml'

****** xocc v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

Attempting to get a license: ap_opencl
Feature available: ap_opencl
INFO: [XOCC 60-1306] Additional information associated with this xocc compile can be found at:
	Reports: /home/jan/workspace/sycl-bugreports/xilinx/typedefs/_x/reports/xSYCL16676224068994425510
	Log files: /home/jan/workspace/sycl-bugreports/xilinx/typedefs/_x/logs/xSYCL16676224068994425510
INFO: [XOCC 60-585] Compiling for hardware emulation target
INFO: [XOCC 60-1316] Initiating connection to rulecheck server, at Thu Oct 10 14:44:57 2019
Running Rule Check Server on port:45569
INFO: [XOCC 60-1315] Creating rulecheck session with output '/home/jan/workspace/sycl-bugreports/xilinx/typedefs/_x/reports/xSYCL16676224068994425510/xocc_compile_xSYCL16676224068994425510_guidance.html', at Thu Oct 10 14:44:58 2019
INFO: [XOCC 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u200_xdma_201830_2/xilinx_u200_xdma_201830_2.xpfm
INFO: [XOCC 60-423]   Target device: xilinx_u200_xdma_201830_2
INFO: [XOCC 60-242] Creating kernel: 'xSYCL16676224068994425510'
ERROR: [XOCC 17-1309] Gcc: #17 0x00007f774215bb6b __libc_start_main /build/glibc-KRRWSm/glibc-2.29/csu/../csu/libc-start.c:308:16
ERROR: [XOCC 60-398] clang failed
ERROR: [XOCC 60-599] Kernel compilation failed to complete
ERROR: [XOCC 60-592] Failed to finish compilation
Option Map File Used: '/opt/xilinx/SDx/2019.1/data/sdx/xocc/optMap.xml'

****** xocc v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

ERROR: [XOCC 60-602] Source file does not exist: /tmp/xSYCL16676224068994425510.xo
ERROR: [XOCC 60-623] Unsupported input file type specified.
rm: cannot remove '/tmp/xSYCL16676224068994425510.xo': No such file or directory
```

agozillon, 2019-10-10:
This may be similar to the previous comment I left on the other issue in that the environment variable should no longer be set. In this case there is also a test I added recently based on this example which should recreate the issue and it passes on my machine (so hopefully it's not a fluke and it works for you as well without the env variable): 
https://github.com/triSYCL/sycl/blob/sycl/unified/next/sycl/test/xocc_tests/simple_tests/kernel_uint_name.cpp