triSYCL/sycl #67: [SYCL] What is the purpose of the reqd_work_group_size extension?

j-stephan, 2019-08-18:
I noticed that there is a Xilinx-specific extension for `reqd_work_group_size`. I'm struggling to grasp the purpose of this extension since the SYCL specification already defines a function attribute in section 6.7. Is the extension simply for convenience or does it have some other implications I'm not aware of?

agozillon, 2019-08-19:
It's purpose is the same as the attribute, except I imagine the attribute can't be used with template parameters in the upstream implementation I suppose, I could be wrong though someone might have wired it appropriately in the compiler for that to work. In either case you can use them hopefully interchangeably, although I'm not sure the patch required for activation of the attribute for SYCL kernels has been pulled in yet. 

Specification aside the extension exists because we wanted to see how easy it would be to implement something that could generically pass information down to our optimization passes without having to extend the Clang frontend much/if at all via adding new attributes/attribute behaviors or keywords etc. Preferably using some kind of Modern C++ method or style. I landed on this type trait style extension after trying a few other things, it works but I imagine there's areas it can be improved upon or better directions. I think for example if someone wanted to extend the annotate attribute in Clang/LLVM to behave better you could pass a lot of arbitrary information down to your optimization passes via C++ wrappers. 

I believe this is useful in our case because:
1) we don't have a lot of resources to put behind adding new attributes for every FPGA extension we have in our FPGA backend 
2) We try to avoid altering the frontend to not have a more complex time when we merge from upstream

We chose the reqd_work_group_size attribute as it wasn't implemented for SYCL kernels until recently and it's pretty important when compiling for FPGA (at least ours anyway) to specify the exact size of your work group at compile time.

keryell, 2019-08-19:
@j-stephan Attributes are bad since it is a C++ extension requiring some compiler support.
I think the few attributes still remaining in the SYCL 1.2.1 are a design error that we should remove, when possible.
So you can see this extension as some experiments for some future versions.

j-stephan, 2019-08-19:
Thank you both, that helped.

closed 2019-08-19 by j-stephan