triSYCL/sycl #69: [SYCL] Invalid record when comparing reference to value

j-stephan, 2019-09-20:
Example code below. This bug isn't present in the Intel compiler (or has been fixed already). After the initial error occurs compilation resumes and produces an executable which doesn't crash.

Command line and subsequent output:

```
$ LC_ALL="C" clang++ -O3 -std=c++2a -fsycl -fsycl-targets=fpga64-xilinx-unknown-sycldevice main.cpp -o invalid_record -lOpenCL -I/opt/xilinx/xrt/include

/home/jan/software/sycl/xilinx/bin/opt: /tmp/main-bee693.o: error: Invalid record (Producer: 'LLVM9.0.0svn' Reader: 'LLVM 9.0.0svn')
warning: Linking two modules of different data layouts: '/opt/xilinx/SDx/2019.1/bin/../lnx64/lib/libspir64-39-hls.bc' is 'e-m:e-i64:64-i128:128-i256:256-i512:512-i1024:1024-i2048:2048-i4096:4096-n8:16:32:64-S128-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024' whereas 'llvm-link' is 'e-i64:64-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024'

warning: Linking two modules of different target triples: /opt/xilinx/SDx/2019.1/bin/../lnx64/lib/libspir64-39-hls.bc' is 'fpga64-xilinx-none' whereas 'llvm-link' is 'spir64'

Invoking xocc Kernel Compilation
xocc: /opt/xilinx/SDx/2019.1/bin/xocc
--target: sw_emu
--platform: xilinx_u200_xdma_201830_2
Compiling kernel: xSYCL9318603472913316893
Outputting file to: /tmp/xSYCL9318603472913316893.xo
Input file is: /tmp/main_kernels-linked.xpirbc
Option Map File Used: '/opt/xilinx/SDx/2019.1/data/sdx/xocc/optMap.xml'

****** xocc v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

Attempting to get a license: ap_opencl
Feature available: ap_opencl
INFO: [XOCC 60-1306] Additional information associated with this xocc compile can be found at:
	Reports: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/xSYCL9318603472913316893
	Log files: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/logs/xSYCL9318603472913316893
INFO: [XOCC 60-585] Compiling for software emulation target
INFO: [XOCC 60-1316] Initiating connection to rulecheck server, at Fri Sep 20 17:24:53 2019
Running Rule Check Server on port:45325
INFO: [XOCC 60-1315] Creating rulecheck session with output '/home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/xSYCL9318603472913316893/xocc_compile_xSYCL9318603472913316893_guidance.html', at Fri Sep 20 17:24:54 2019
INFO: [XOCC 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u200_xdma_201830_2/xilinx_u200_xdma_201830_2.xpfm
INFO: [XOCC 60-423]   Target device: xilinx_u200_xdma_201830_2
INFO: [XOCC 60-242] Creating kernel: 'xSYCL9318603472913316893'
INFO: [XOCC 60-594] Finished kernel compilation
INFO: [XOCC 60-586] Created /tmp/xSYCL9318603472913316893.xo
INFO: [XOCC 60-791] Total elapsed time: 0h 0m 9s
Option Map File Used: '/opt/xilinx/SDx/2019.1/data/sdx/xocc/optMap.xml'

****** xocc v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

Attempting to get a license: ap_opencl
Feature available: ap_opencl
INFO: [XOCC 60-1306] Additional information associated with this xocc link can be found at:
	Reports: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/link
	Log files: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/logs/link
INFO: [XOCC 60-629] Linking for software emulation target
INFO: [XOCC 60-1316] Initiating connection to rulecheck server, at Fri Sep 20 17:25:04 2019
Running Rule Check Server on port:45591
INFO: [XOCC 60-1315] Creating rulecheck session with output '/home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/link/xocc_link_main-261146_guidance.html', at Fri Sep 20 17:25:05 2019
INFO: [XOCC 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u200_xdma_201830_2/xilinx_u200_xdma_201830_2.xpfm
INFO: [XOCC 60-423]   Target device: xilinx_u200_xdma_201830_2
INFO: [XOCC 60-586] Created /tmp/main-261146.out
INFO: [XOCC 60-1307] Run completed. Additional information can be found in:
	Guidance: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/link/xocc_link_main-261146_guidance.html
	Steps Log File: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/logs/link/link.steps.log

INFO: [XOCC 60-791] Total elapsed time: 0h 0m 14s
```

Example code:

```c++
#include <CL/sycl.hpp>

struct kernel;

auto cmp(std::size_t& large, std::size_t small)
{
    auto val = (large > small) ? small : large;
    return val;
}

auto main() -> int
{
    auto queue = cl::sycl::queue{};

    auto s_buf = cl::sycl::buffer<std::size_t, 3>{
                    cl::sycl::range<3>{42, 42, 42}};

    queue.submit([&](cl::sycl::handler& cgh)
    {
        auto s = s_buf.get_access<cl::sycl::access::mode::read_write>(cgh);

        cgh.single_task<kernel>([=]()
        {
            const auto id = cl::sycl::id<3>{0, 0, 0};
            s[id] = cmp(s[id], 40);
        });
                                                
    });
    queue.wait();

    return 0;
}
```

j-stephan, 2019-09-20:
Follow-up: I can work around this issue in `sw_emu` mode if I copy `large` to a function-local variable:

```c++
auto cmp(std::size_t& large, std::size_t small)
{
    auto large_copy = large;
    auto val = (large_copy > small) ? small : large_copy;
    return val;
}
```

However, this will cause an ICE in `hw_emu`:

```
warning: Linking two modules of different data layouts: '/opt/xilinx/SDx/2019.1/bin/../lnx64/lib/libspir64-39-hls.bc' is 'e-m:e-i64:64-i128:128-i256:256-i512:512-i1024:1024-i2048:2048-i4096:4096-n8:16:32:64-S128-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024' whereas 'llvm-link' is 'e-i64:64-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024'

warning: Linking two modules of different target triples: /opt/xilinx/SDx/2019.1/bin/../lnx64/lib/libspir64-39-hls.bc' is 'fpga64-xilinx-none' whereas 'llvm-link' is 'spir64'

Invoking xocc Kernel Compilation
xocc: /opt/xilinx/SDx/2019.1/bin/xocc
--target: hw_emu
--platform: xilinx_u200_xdma_201830_2
Compiling kernel: xSYCL9318603472913316893
Outputting file to: /tmp/xSYCL9318603472913316893.xo
Input file is: /tmp/main_kernels-linked.xpirbc
Option Map File Used: '/opt/xilinx/SDx/2019.1/data/sdx/xocc/optMap.xml'

****** xocc v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

Attempting to get a license: ap_opencl
Feature available: ap_opencl
INFO: [XOCC 60-1306] Additional information associated with this xocc compile can be found at:
	Reports: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/xSYCL9318603472913316893
	Log files: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/logs/xSYCL9318603472913316893
INFO: [XOCC 60-585] Compiling for hardware emulation target
INFO: [XOCC 60-1316] Initiating connection to rulecheck server, at Fri Sep 20 18:19:22 2019
Running Rule Check Server on port:36007
INFO: [XOCC 60-1315] Creating rulecheck session with output '/home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/xSYCL9318603472913316893/xocc_compile_xSYCL9318603472913316893_guidance.html', at Fri Sep 20 18:19:23 2019
INFO: [XOCC 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u200_xdma_201830_2/xilinx_u200_xdma_201830_2.xpfm
INFO: [XOCC 60-423]   Target device: xilinx_u200_xdma_201830_2
INFO: [XOCC 60-242] Creating kernel: 'xSYCL9318603472913316893'
ERROR: [XOCC 17-1309] Gcc: #17 0x00007f28dce65b6b __libc_start_main /build/glibc-KRRWSm/glibc-2.29/csu/../csu/libc-start.c:308:16
ERROR: [XOCC 60-398] clang failed
ERROR: [XOCC 60-599] Kernel compilation failed to complete
ERROR: [XOCC 60-592] Failed to finish compilation
Option Map File Used: '/opt/xilinx/SDx/2019.1/data/sdx/xocc/optMap.xml'

****** xocc v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

ERROR: [XOCC 60-632] XO validation failed: ERROR: [XOCC 17-70] Application Exception: The kernel 'xSYCL9318603472913316893' in '/tmp/xSYCL9318603472913316893.xo' is not valid for the specified target 'hw_emu'. The valid target: sw_emu

/usr/bin/ld: /tmp/main-515508.o: file not recognized: file truncated
clang-9: error: sycl-link-xocc command failed with exit code 255 (use -v to see invocation)
clang-9: error: linker command failed with exit code 1 (use -v to see invocation)

agozillon, 2019-10-08:
I think this is now fixed and close able in the sycl/unified/next branch. Feel free to give it a try when you get a spare minute and close this issue and/or close the issue and open it again if you find a problem!

j-stephan, 2019-10-10:
The `sw_emu` issue is fixed. Unfortunately the source code from the original example (without the local copy) causes an ICE in `hw_emu` mode:

```
[INFO] Could not find v++ executable in /opt/xilinx/SDx/2019.1/bin
[INFO] Try with xocc...
warning: Linking two modules of different data layouts: '/opt/xilinx/SDx/2019.1/bin/../lnx64/lib/libspir64-39-hls.bc' is 'e-m:e-i64:64-i128:128-i256:256-i512:512-i1024:1024-i2048:2048-i4096:4096-n8:16:32:64-S128-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024' whereas 'llvm-link' is 'e-i64:64-v16:16-v24:32-v32:32-v48:64-v96:128-v192:256-v256:256-v512:512-v1024:1024'

warning: Linking two modules of different target triples: /opt/xilinx/SDx/2019.1/bin/../lnx64/lib/libspir64-39-hls.bc' is 'fpga64-xilinx-none' whereas 'llvm-link' is 'spir64'

Invoking Kernel Compilation with /opt/xilinx/SDx/2019.1/bin/xocc
--target: hw_emu
--platform: xilinx_u200_xdma_201830_2
Compiling kernel: xSYCL9318603472913316893
Outputting file to: /tmp/xSYCL9318603472913316893.xo
Input file is: /tmp/main_kernels-linked.xpirbc
Option Map File Used: '/opt/xilinx/SDx/2019.1/data/sdx/xocc/optMap.xml'

****** xocc v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

Attempting to get a license: ap_opencl
Feature available: ap_opencl
INFO: [XOCC 60-1306] Additional information associated with this xocc compile can be found at:
	Reports: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/xSYCL9318603472913316893
	Log files: /home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/logs/xSYCL9318603472913316893
INFO: [XOCC 60-585] Compiling for hardware emulation target
INFO: [XOCC 60-1316] Initiating connection to rulecheck server, at Thu Oct 10 14:38:25 2019
Running Rule Check Server on port:41357
INFO: [XOCC 60-1315] Creating rulecheck session with output '/home/jan/workspace/sycl-bugreports/xilinx/invalid_record/_x/reports/xSYCL9318603472913316893/xocc_compile_xSYCL9318603472913316893_guidance.html', at Thu Oct 10 14:38:26 2019
INFO: [XOCC 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u200_xdma_201830_2/xilinx_u200_xdma_201830_2.xpfm
INFO: [XOCC 60-423]   Target device: xilinx_u200_xdma_201830_2
INFO: [XOCC 60-242] Creating kernel: 'xSYCL9318603472913316893'
ERROR: [XOCC 17-1309] Gcc: #17 0x00007fa5f3e8eb6b __libc_start_main /build/glibc-KRRWSm/glibc-2.29/csu/../csu/libc-start.c:308:16
ERROR: [XOCC 60-398] clang failed
ERROR: [XOCC 60-599] Kernel compilation failed to complete
ERROR: [XOCC 60-592] Failed to finish compilation
Option Map File Used: '/opt/xilinx/SDx/2019.1/data/sdx/xocc/optMap.xml'

****** xocc v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

ERROR: [XOCC 60-602] Source file does not exist: /tmp/xSYCL9318603472913316893.xo
ERROR: [XOCC 60-623] Unsupported input file type specified.
rm: cannot remove '/tmp/xSYCL9318603472913316893.xo': No such file or directory
```

agozillon, 2019-10-10:
I probably should have mentioned but the DISBLE_AS_INFER environment variable should no longer be set or you're likely to have worse behavior than previously: https://github.com/triSYCL/sycl/pull/71/commits/5e48bc5a396ae44d2eb4eb8e7fa44b1526b73c9e

j-stephan, 2019-10-10:
Okay, that did the trick. Should I discard `DISABLE_INFER_AS` completely or is it still required in some cases?