triSYCL/sycl #78: [SYCL] hls's libspir contains unoptimized IR and some buggy IR

Ralender, 2020-09-14:
This is an HLS bug not a trisycl bug

the library is located at $XILINX_ROOT/Vitis/2020.1/lnx64/lib/libspir64-39-hls.bc and is shipped as part of Vitis
it isn't optimized, which increases build time and it contains some buggy functions like
`
; Function Attrs: alwaysinline nounwind
define spir_func float @erfc_impl(float %x) #29 {
  %retval = alloca float, align 4
  %x.addr = alloca float, align 4
  store float %x, float* %x.addr, align 4
  call void @llvm.trap()
  unreachable

1:                                                ; No predecessors!
  %2 = load float, float* %retval, align 4
  ret float %2
}
`

keryell, 2020-09-14:
@aisoard @yu810226 any idea about how this code is generated? It looks like rubbish code...

yu810226, 2020-09-22:
Seems to be completely useless code.

keryell, 2020-09-23:
Any code coverage tools? :-)

keryell, 2020-09-23:
@yu810226 I am curious about how this got generated. I let you fix this in HLS?