triSYCL/sycl #81: Full compilation fails

keryell, 2020-09-26:
All the full compilation of `sycl/unified/next`, `Ralender/Merge2` and `sycl/unified/next`+`Ralender/Merge2` fail with various mileage.
@Ralender Could you first merge `sycl/unified/next` into your latest `Ralender/Merge2` and test the following compilation recipe to be sure we are not breaking the original big picture?
```bash
# Pick some place where SYCL has to be compiled:
export SYCL_HOME=~/sycl_workspace
mkdir $SYCL_HOME
cd $SYCL_HOME
git clone --branch sycl/unified/next git@github.com:triSYCL/sycl.git llvm
python $SYCL_HOME/llvm/buildbot/configure.py
python $SYCL_HOME/llvm/buildbot/compile.py
```

Ralender, 2020-09-28:
With the latest version of the PR I linked the following recipe compiles.

keryell, 2020-09-29:
Thanks, it works with the new branch now.

closed 2020-09-29 by keryell