triSYCL/sycl #83: llvm/sycl/test/xocc_tests/disabled/edge_detection example does not work

keryell, 2020-09-29:
The example stated in `sycl/doc/GettingStartedAlveo.md`:
- sw_emu:
  It compiles and runs (indefinitely?) while displaying
  ```
  EXE: /home/rkeryell/sycl_workspace/llvm/sycl/test/xocc_tests/disabled/edge_detection/edge_detection
  [XRT] WARNING: unaligned host pointer '0x7fe590429040' detected, this leads to extra memcpy
  /opt/xilinx/Vitis/2020.1/data/emulation/unified/cpu_em/generic_pcie/model/genericpciemodel: symbol lookup error: /home/rkeryell/sycl_workspace/llvm/sycl/test/xocc_tests/disabled/edge_detection/.run/907981/sw_emu/device0/binary_0/dltmp: undefined symbol: llvm.sideeffect
  ```
  Is there an issue because the old internal LLVM chokes on newer `llvm.sideeffect`?

- hw_emu:
  ```
  INFO: [v++ 60-585] Compiling for hardware emulation target
  INFO: [v++ 60-423]   Target device: xilinx_u200_xdma_201830_2
  INFO: [v++ 60-242] Creating kernel: 'xSYCL13730816572338433693'
  ERROR: [v++ 60-300] Failed to build kernel(ip) xSYCL13730816572338433693, see log for details: /home/rkeryell/sycl_workspace/llvm/sycl/test/xocc_tests/disabled/edge_detection/_x/sycl-xocc.Nytewv_xSYCL13730816572338433693/xSYCL13730816572338433693/vitis_hls.log
  ERROR: [v++ 60-599] Kernel compilation failed to complete
  ERROR: [v++ 60-592] Failed to finish compilation
  INFO: [v++ 60-1653] Closing dispatch client.
  clang-12: error: sycl-link-xocc command failed with exit code 1 (use -v to see invocation)
  ```

- hw:
  ```
  INFO: [v++ 60-242] Creating kernel: 'xSYCL13730816572338433693'
  ERROR: [v++ 60-300] Failed to build kernel(ip) xSYCL13730816572338433693, see log for details: /home/rkeryell/sycl_workspace/llvm/sycl/test/xocc_tests/disabled/edge_detection/_x/sycl-xocc.03eRpw_xSYCL13730816572338433693/xSYCL13730816572338433693/vitis_hls.log
  ERROR: [v++ 60-599] Kernel compilation failed to complete
  ERROR: [v++ 60-592] Failed to finish compilation
  INFO: [v++ 60-1653] Closing dispatch client.
  clang-12: error: sycl-link-xocc command failed with exit code 1 (use -v to see invocation)
  rkeryell@rk-xsj:~/sycl_workspace/llvm/sycl/test/xocc_tests/disabled/edge_detection (sycl/unified/next)$ m /home/rkeryell/sycl_workspace/llvm/sycl/test/xocc_tests/disabled/edge_detection/_x/sycl-xocc.03eRpw_xSYCL13730816572338433693/xSYCL13730816572338433693/vitis_hls.log
  ```

Ralender, 2020-09-30:
I didn't experience any issues in sw_emu but I had the same issue in hw_emu and the linked PR fixes this.

Ralender, 2020-10-01:
After some more testing, the issue in sw_emu is still present in on Vitis 2020.1 not in 2020.2.