triSYCL/sycl #84: SYCL Vitis 2020.1 tests do not work out-of-the-box

keryell, 2020-09-29:
I have updated the recipe to be independent from `make` https://github.com/triSYCL/sycl/pull/80/commits/b26fd4a9792d8ee9e72be9ef1e992d082bcf25d6
There are still some issues with the fact we compile a lot of stuff (like level-0) while injecting Xilinx code relying on C++20 but the global library is not compiled with C++20. Put C++20 in the library compilation flags?

Ralender, 2020-09-30:
the sycl library is already build in C++20 mode.
could you provide specifics about the issue your a seeing.

keryell, 2020-09-30:
Probably I do not really understand how to run the tests in a portable way, like I started to detail in the doc.
Following the recipe
```
cd $SYCL_HOME/llvm/build
rkeryell@rk-xsj:~/sycl_workspace/llvm/build (sycl/unified/next)$   export XCL_EMULATION_MODE=sw_emu
rkeryell@rk-xsj:~/sycl_workspace/llvm/build (sycl/unified/next)$   cmake --build . --parallel `nproc` --target check-sycl-xocc-jmax
[...]
FAILED: tools/sycl/unittests/queue/CMakeFiles/QueueTests.dir/wait.cpp.o 
/usr/bin/c++  -DCL_TARGET_OPENCL_VERSION=220 -DGTEST_HAS_TR1_TUPLE=0 -D_DEBUG -D_GLIBCXX_ASSERTIONS=1 -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -D__SYCL_BUILD_SYCL_DLL -D__SYCL_XILINX_ONLY__ -Itools/sycl/unittests/queue -I/home/rkeryell/sycl_workspace/llvm/sycl/unittests/queue -I/usr/include/libxml2 -Iinclude -I/home/rkeryell/sycl_workspace/llvm/llvm/include -I/home/rkeryell/sycl_workspace/llvm/sycl/unittests/queue/SYSTEM -I/home/rkeryell/sycl_workspace/llvm/sycl/include -I/home/rkeryell/sycl_workspace/llvm/sycl/source -I/home/rkeryell/sycl_workspace/llvm/sycl/unittests -I/home/rkeryell/sycl_workspace/llvm/llvm/utils/unittest/googletest/include -I/home/rkeryell/sycl_workspace/llvm/llvm/utils/unittest/googlemock/include -Itools/sycl/OpenCL/inc -isystem /opt/xilinx/xrt/include -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wsuggest-override -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -Wall -Wextra -Wno-deprecated-declarations -Werror -O3     -Wno-variadic-macros -UNDEBUG -Wno-suggest-override -std=c++14 -MD -MT tools/sycl/unittests/queue/CMakeFiles/QueueTests.dir/wait.cpp.o -MF tools/sycl/unittests/queue/CMakeFiles/QueueTests.dir/wait.cpp.o.d -o tools/sycl/unittests/queue/CMakeFiles/QueueTests.dir/wait.cpp.o -c /home/rkeryell/sycl_workspace/llvm/sycl/unittests/queue/wait.cpp
In file included from /home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl/handler.hpp:31,
                 from /home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl/ONEAPI/reduction.hpp:13,
                 from /home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl.hpp:16,
                 from /home/rkeryell/sycl_workspace/llvm/sycl/unittests/queue/wait.cpp:9:
/home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl/xilinx/fpga/kernel_properties.hpp:27:11: error: nested namespace definitions only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Werror=pedantic]
   27 | namespace sycl::xilinx {
      |           ^~~~
In file included from /home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl/xilinx/fpga.hpp:18,
                 from /home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl.hpp:50,
                 from /home/rkeryell/sycl_workspace/llvm/sycl/unittests/queue/wait.cpp:9:
/home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl/xilinx/fpga/opt_decorate_func.hpp:29:11: error: nested namespace definitions only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Werror=pedantic]
   29 | namespace sycl::xilinx {
      |           ^~~~
In file included from /home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl/xilinx/fpga.hpp:19,
                 from /home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl.hpp:50,
                 from /home/rkeryell/sycl_workspace/llvm/sycl/unittests/queue/wait.cpp:9:
/home/rkeryell/sycl_workspace/llvm/sycl/include/CL/sycl/xilinx/fpga/partition_array.hpp:31:11: error: nested namespace definitions only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Werror=pedantic]
[...]
```

keryell, 2020-09-30:
Surprisingly, this works rather well:
```bash
python $SYCL_HOME/llvm/buildbot/check.py
args:Namespace(base_branch=None, branch=None, build_number=None, builder_dir=None, obj_dir=None, pr_number=None, src_dir=None, test_suite='check-all')
[Cmake Command]: cmake --build /home/rkeryell/sycl_workspace/llvm/build -- check-all -j 12
Failed Tests (4):
  LLVM-Unit :: ADT/./ADTTests/TripleTest.Normalization
  Clang :: Driver/sycl-intelfpga-static-lib.cpp
  Clang :: SemaSYCL/intel-fpga-mem-builtin.cpp
  Clang :: SemaSYCL/intel-fpga-reg.cpp
[...]
Testing Time: 364.19s
  Unsupported      : 17591
  Passed           : 48196
  Expectedly Failed:    85
  Failed           :     4
FAILED: CMakeFiles/check-all 
```

keryell, 2020-09-30:
OK, I found the problem in `sycl/cmake/modules/AddSYCLUnitTest.cmake` where there is C++14 instead C++20.
I can fix this.

keryell, 2020-09-30:
I solved the C++20 issue with https://github.com/triSYCL/sycl/commit/9a3f148e86000e171a239b268cd9d335d3da0df6 but there is still an interesting issue.
When running the tests several times, the temporary files left over by `v++` are interpreted by `LIT` as potential test files that are unresolved! :-/
```
[...]
UNRESOLVED: SYCL-XOCC :: simple_tests/_x/sycl-xocc.6jcZs8_xSYCL14115975588455758984/xSYCL14115975588455758984/xSYCL14115975588455758984/solution/impl/misc/drivers/xSYCL14115975588455758984_v1_0/src/xxsycl14115975588455758984_linux.c (418 of 434)
******************** TEST 'SYCL-XOCC :: simple_tests/_x/sycl-xocc.6jcZs8_xSYCL14115975588455758984/xSYCL14115975588455758984/xSYCL14115975588455758984/solution/impl/misc/drivers/xSYCL14115975588455758984_v1_0/src/xxsycl14115975588455758984_linux.c' FAILED ********************
Test has no 'RUN:' line
********************
UNRESOLVED: SYCL-XOCC :: simple_tests/_x/sycl-xocc.6jcZs8_xSYCL14115975588455758984/xSYCL14115975588455758984/xSYCL14115975588455758984/solution/impl/misc/drivers/xSYCL14115975588455758984_v1_0/src/xxsycl14115975588455758984_sinit.c (419 of 434)
******************** TEST 'SYCL-XOCC :: simple_tests/_x/sycl-xocc.6jcZs8_xSYCL14115975588455758984/xSYCL14115975588455758984/xSYCL14115975588455758984/solution/impl/misc/drivers/xSYCL14115975588455758984_v1_0/src/xxsycl14115975588455758984_sinit.c' FAILED ********************
Test has no 'RUN:' line
********************
********************
Unresolved Tests (419):
  SYCL-XOCC :: simple_tests/_x/link/int/behav_waveform/xsim/xsim.dir/emu_wrapper_behav/obj/xsim_6.c
  SYCL-XOCC :: simple_tests/_x/link/int/behav_waveform/xsim/xsim.dir/emu_wrapper_behav/obj/xsim_7.c
[...]
  SYCL-XOCC :: simple_tests/_x/sycl-xocc.6jcZs8_xSYCL14115975588455758984/xSYCL14115975588455758984/xSYCL14115975588455758984/solution/impl/misc/drivers/xSYCL14115975588455758984_v1_0/src/xxsycl14115975588455758984_linux.c
  SYCL-XOCC :: simple_tests/_x/sycl-xocc.6jcZs8_xSYCL14115975588455758984/xSYCL14115975588455758984/xSYCL14115975588455758984/solution/impl/misc/drivers/xSYCL14115975588455758984_v1_0/src/xxsycl14115975588455758984_sinit.c

Testing Time: 298.54s
  Passed    :  15
  Unresolved: 419

3 warning(s) in tests
FAILED: tools/sycl/test/CMakeFiles/check-sycl-xocc-j4 
cd /home/rkeryell/sycl_workspace/llvm/build/tools/sycl/test && /usr/bin/python3.8 /home/rkeryell/sycl_workspace/llvm/build/./bin/llvm-lit -v -sv --param XOCC=only /home/rkeryell/sycl_workspace/llvm/build/tools/sycl/test -j\ 4
ninja: build stopped: subcommand failed.
```

closed 2020-10-01 by keryell