triSYCL/sycl #103: sycl-xocc: Wildcard in script not working correctly

j-stephan, 2020-12-03:
The wildcards in the lines below fail for me after the upgrade to Vitis 2020.2. I have to manually edit the `sycl-xocc` script and replace `2*` with `2020.2`.

https://github.com/triSYCL/sycl/blob/7820e5e2cf738fcb5944005609a781363c1a0e4b/sycl/tools/sycl-xocc/bin/sycl-xocc#L97
https://github.com/triSYCL/sycl/blob/7820e5e2cf738fcb5944005609a781363c1a0e4b/sycl/tools/sycl-xocc/bin/sycl-xocc#L105

If I don't do this I end up with the following error:

```bash
++ pwd
+ CWD=/home/jan/workspace/zoo/alpaka/build/test/integ/axpy
+ export VERBOSE=1
+ VERBOSE=1
+ [[ -z /home/jan/software/xilinx/Vitis/2020.2/bin ]]
+ [[ -z /home/jan/sycl-workspace/SplitCompilation/sycl/build/bin ]]
+ [[ -z axpy.cpp-2f6f39 ]]
+ [[ -z /tmp/axpy.cpp-2f6f39.txt ]]
+ [[ -z /tmp ]]
+ [[ -z /tmp/axpy-671ac1.bc ]]
+ [[ -z /tmp/sycl-xocc-args2712-3d2a4b ]]
+ [[ -z /tmp/sycl-xocc-args2716-4c91c5 ]]
+ [[ -z sw_emu ]]
+ SDX_BIN_PATH_DIR=/home/jan/software/xilinx/Vitis/2020.2/bin
+ [[ ! -d /home/jan/software/xilinx/Vitis/2020.2/bin ]]
+ SDX_LIB_SPIR=/home/jan/software/xilinx/Vitis/2020.2/bin/../lnx64/lib/libspir64-39-hls.bc
+ [[ ! -f /home/jan/software/xilinx/Vitis/2020.2/bin/../lnx64/lib/libspir64-39-hls.bc ]]
+ SDX_LIB_SPIR='/home/jan/software/xilinx/Vitis/2020.2/bin/../../../Vitis_HLS/2*/lnx64/lib/libspir64-39-hls.bc'
+ [[ ! -f /home/jan/software/xilinx/Vitis/2020.2/bin/../../../Vitis_HLS/2*/lnx64/lib/libspir64-39-hls.bc ]]
+ usage 9 'could not find SDx'\''s libspir64-39-hls.bc'
+ echo /home/jan/sycl-workspace/SplitCompilation/sycl/build/bin/sycl-xocc: error: could not find 'SDx'\''s' libspir64-39-hls.bc
/home/jan/sycl-workspace/SplitCompilation/sycl/build/bin/sycl-xocc: error: could not find SDx's libspir64-39-hls.bc
+ exit 9
clang-12: error: sycl-link-xocc command failed with exit code 9 (use -v to see invocation)
```

I first assumed that this was because my default shell is `zsh` (version 5.8) but it also occurs if I run it from `bash` (version 5.0.17).

keryell, 2021-05-25:
I think this has been solved now. @Ralender ?

j-stephan, 2021-12-10:
So far I haven't encountered this error again. I guess this can be closed.

closed 2021-12-10 by j-stephan