triSYCL/sycl #123: Feature support and general questions

gogo2, 2021-05-27:
I am recently evaluating triSYCL/sycl capabilities and its suitability to be used in something real.

Regarding the core features of SYCL - I see that the basics are well supported and I didn't have any major issues compiling and even running our tests and bit-more-complex-than-matmul projects on real U250.

But there are two fpga-kind-of-important features that I'm not sure about, namely:
* ##### DDR banks
I tried to specify DDR banks for accessors in a simple VecAdd example.
The feature itself works, correct banks are used, but I see ~4x performance drop compared to the default DDR[0] version.
Do you have any idea what can be the cause?

* ##### reqd_work_group_size
I saw the test of this property in the `disabled` directory, but it doesn't explicitly say that the functionality is broken.
Is this property working and can it be used? There is a note in the test that it requires an additional compilation flag, but does not specify. I tried to _just use it_ but it seems to have no effect.

#### Perspectives for the future
I asked you during IWOCL about the future of the project, and @keryell said that it's not going anywhere, but do you have any predictions on how it will evolve in terms of feature support (SYCL specific like hierarchical kernels and FPGA specific like HBM banks, or more HLS pragmas equivalents), stability and performance?

Also, If you are interested, I could provide you with some insights about the usage and test results of triSYCL/sycl in my team.

keryell, 2021-05-28:
The DDR banks for accessors are not well tested. We need to investigate. Please share your code sample.
Good point for the tests, we need to clean up the test directory.
For the future we are in the process of moving from the OpenCL backend to the HLS backend to benefit potentially from quite more control and performance, starting with https://github.com/triSYCL/sycl/pull/121
The problem is to get the full picture working while Vitis has not been designed for something wild like SYCL and there is no official support from the product teams, so the ride is quite bumpy... But we keep trying. :-)
I am unsure about what you mean by hierarchical kernels. Hierarchical parallelism? On FPGA, we can probably handle them in a similar way to the executor in https://github.com/triSYCL/path_tracer/blob/main/include/render.hpp#L111 because they might mo be as important as GPU for example.
"@keryell said that it's not going anywhere,": if I said this, it was probably a side effect of my approximate English I/O subsystem. ;-) Just that for now this is a small research project at Xilinx.
But globally it should evolve towards more features, stability and performance, not at the pace I would dream of...
We are also working on our Versal ACAP CGRA platform, which dilutes the efforts. Stay tuned.
In any case, please provide some feedback and code samples from your team!

gogo2, 2021-05-28:
Ok, so here I attach two versions of vectro addition:
[trisycl_vec_add_tests.zip](https://github.com/triSYCL/sycl/files/6559784/trisycl_vec_add_tests.zip)
The DDR banks version executes about 4 times longer that the default one.

I use:
* Vitis 2020.2
* XRT 2.9.317 on build machine and 2.5.309 on execution machine
* Alveo U250 with XDMA 201830.2 shell
* Ubuntu 18.04
* triSYCL/sycl from current sycl/unified/master branch

gogo2, 2021-05-28:
> I am unsure about what you mean by hierarchical kernels. Hierarchical parallelism? On FPGA, we can probably handle them...

It's not like I need this particular feature now. What I meant was the general ability to grab an existing SYCL kernel, build with triSYCL/sycl, see the outcome and then apply FPGA-specific optimizations and features.

I will soon collect a set of information about our use cases and share.

gogo2, 2021-06-23:
Sorry for the late response, last few weeks have been especially busy.

So a little bit more about the case:
- The projects at the moment consists of ~600 device compiled LOC in 4 kernels and auxiliary functions
- About third to half of the above is branch heavy code
- Alveo run was about 36 times slower than hipSYCL/OpenMP on midend 6-core CPU (both compiled from exactly the same source code)
- Compiler placed only one set of kernels consuming ~14% of LUTs, which I suppose is the correct default behavior, but that is why I asked about `reqd_work_group_size`, if there is any way to use it for Xilinx target.

Looking forward to the HLS backend being ready for testing!

lforg37, 2021-07-12:
Hello, 
Thank you for your test file !
The HLS backend has been merged in sycl/unified/next, and with it some improvement of the memory assignment properties.
Now the behavior is to create  one m_axi controller for each memory bench having a buffer associated to it, which was not the case earlier.
I've tested  it with the file you provided, and we now have an improvement when using a different memory bank for each buffer.
The IR annotation we use to specify this is not understood by the OpenCL backend, and therefore the memory assignment works only with the HLS backend.

> Compiler placed only one set of kernels consuming ~14% of LUTs, which I suppose is the correct default behavior, but that is why I asked about reqd_work_group_size, if there is any way to use it for Xilinx target.

We plan to allow the user to choose how many time the kernels should be placed, but the interface for this is still being discussed.

closed 2022-05-18 by gogo2