triSYCL/sycl #166: HLS compilation flow redefines `__WINT_TYPE__` to `int`

j-stephan, 2022-02-23:
Reproducer which shows that this breaks the standard include files:

```
$ cat foo.cpp
#include <cwchar>

$ clang++ -std=c++20 -fsycl -fsycl-targets=fpga64_hls_hw_emu -c foo.cpp
In file included from foo.cpp:1:
In file included from /usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/cwchar:44:
/usr/include/wchar.h:61:22: error: typedef redefinition with different types ('unsigned int' vs 'int')
typedef unsigned int wint_t;
                     ^
/home/stepha27/sycl-workspace/sycl-build/lib/clang/14.0.0/include/stddef.h:116:23: note: previous definition is here
typedef __WINT_TYPE__ wint_t;
                      ^
1 error generated.

# Generic target for comparison
$ clang++ -std=c++20 -fsycl -c foo.cpp
```

It seems that the HLS flow redefines `__WINT_TYPE__` to `int` although it should be `unsigned int` (`bar.cpp` is empty):

```
$ clang++ -std=c++20 -fsycl -dM -E -c bar.cpp | grep WINT_TYPE
#define __WINT_TYPE__ unsigned int
#define __WINT_TYPE__ unsigned int

$ clang++ -std=c++20 -fsycl -fsycl-targets=fpga64_hls_hw_emu -dM -E -c foo.cpp | grep WINT_TYPE
#define __WINT_TYPE__ int
#define __WINT_TYPE__ unsigned int
```

The above has been produced with the current `sycl/unified/next` which was compiled by GCC 11.2.0 on a CentOS-based cluster.

Ralender, 2022-02-24:
Thanks you for the bug report. I added a patch that fixes the issue on my side.

closed 2022-02-28 by keryell

j-stephan, 2022-03-01:
Thanks for addressing this so quickly. This has only been merged into `sycl/unified/master` though, right? Is the divergence between `master` and `next` intentional?

keryell, 2022-03-01:
> Thanks for addressing this so quickly. This has only been merged into `sycl/unified/master` though, right? Is the divergence between `master` and `next` intentional?

Oh no! Thank you for catching this. I did not check carefully enough the PR branch destination before merging. :-(
I am going to merge also in `sycl/unified/next` to avoid rewriting public history.
Thank you for having your eyes wide open on the project! :-)