triSYCL/sycl #168: `fpga_hls_hw_emu` fails in `vpl` step

j-stephan, 2022-03-02:
I'm on the current `sycl/unified/next` branch and using Vitis 2020.2 with a `xilinx_u200_xdma_201830_2` target. Compiling the `single_task_vector_add` test case works fine but it fails once `vpl` is called:

```
$ clang++ --gcc-toolchain=/trinity/shared/pkg/compiler/gcc/11.2.0 -std=c++20 -fsycl -fsycl-targets=fpga64_hls_hw_emu single_task_vector_add.cpp

[Snipped - Previous messages in attached log file]

****** vpl v2020.2 (64-bit)
  **** SW Build (by xbuild) on 2020-11-18-05:13:29
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file '/tmp/single_task_vector_add-47f9b65zz08txn/vxx_link_tmp/link/int/kernel_info.dat'.
INFO: [VPL 74-74] Compiler Version string: 2020.2
INFO: [VPL 60-423]   Target device: xilinx_u200_xdma_201830_2
INFO: [VPL 60-1032] Extracting hardware platform to /tmp/single_task_vector_add-47f9b65zz08txn/vxx_link_tmp/link/vivado/vpl/.local/hw_platform
WARNING: /trinity/shared/pkg/devel/sdaccel/Vitis/2020.2/tps/lnx64/jre9.0.4 does not exist.
[11:04:50] Run vpl: Step create_project: Started
Creating Vivado project.
[11:04:53] Run vpl: Step create_project: Completed
[11:04:53] Run vpl: Step create_bd: Started
[11:05:37] Run vpl: Step create_bd: Completed
[11:05:37] Run vpl: Step update_bd: Started
[11:05:38] Run vpl: Step update_bd: Completed
[11:05:38] Run vpl: Step generate_target: Started
[11:05:55] Run vpl: Step generate_target: Completed
[11:05:55] Run vpl: Step config_hw_emu.gen_scripts: Started
[11:06:14] Run vpl: Step config_hw_emu.gen_scripts: Completed
[11:06:14] Run vpl: Step config_hw_emu.compile: Started
[11:06:18] Run vpl: Step config_hw_emu.compile: Failed
[11:06:22] Run vpl: FINISHED. Run Status: config_hw_emu.compile ERROR

===>The following messages were generated while processing /tmp/single_task_vector_add-47f9b65zz08txn/vxx_link_tmp/link/vivado/vpl/prj/prj.sim/sim_1/behav_waveform/xsim :
ERROR: [VPL 43-3410] Failed to compile one of the generated C files.
Please recompile with "-mt off -v 1" switch to identify which design unit failed
.
ERROR: [VPL 60-773] In '/tmp/single_task_vector_add-47f9b65zz08txn/vxx_link_tmp/link/vivado/vpl/vivado.log', caught Tcl error:  xsc -c --gcc_compile_options -DBOOST_SYSTEM_NO_DEPRECATED --gcc_compile_options -I/trinity/shared/pkg/devel/sdaccel/Vivado/2020.2/data/simmodels/xsim/2020.2/lnx64/6.2.0/ext/protobuf/include --gcc_compile_options -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_embedded_scheduler_sw_0_0/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_embedded_scheduler_sw_0_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_1_0/sim_tlm/common -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_1_0/sim_tlm/axi_crossbar/systemc_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_1_0/sim_tlm/top -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_1_0/sim_tlm/ddr_model/cpp_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_1_0/sim_tlm/ddr_model/systemc_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_1_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_xdma_0_0/src/common -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_xdma_0_0/src/cpp_src -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_xdma_0_0/src/sysc_src -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_xdma_0_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_xtlm_simple_intercon_0_0/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_xtlm_simple_intercon_0_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/ip/ip_0/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/ip/ip_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/ip/ip_1/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/ip/ip_1/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/ip/ip_2/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/ip/ip_2/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/ip/ip_3/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/ip/ip_3/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/bd_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_memory_subsystem_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_0_0/sim_tlm/common -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_0_0/sim_tlm/axi_crossbar/systemc_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_0_0/sim_tlm/top -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_0_0/sim_tlm/ddr_model/cpp_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_0_0/sim_tlm/ddr_model/systemc_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_0_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_2_0/sim_tlm/common -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_2_0/sim_tlm/axi_crossbar/systemc_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_2_0/sim_tlm/top -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_2_0/sim_tlm/ddr_model/cpp_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_2_0/sim_tlm/ddr_model/systemc_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_2_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_3_0/sim_tlm/common -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_3_0/sim_tlm/axi_crossbar/systemc_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_3_0/sim_tlm/top -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_3_0/sim_tlm/ddr_model/cpp_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_3_0/sim_tlm/ddr_model/systemc_srcs -I../../../../prj.ip_user_files/bd/emu/ip/emu_sim_ddr_3_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_icn_pass_0_0/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_icn_pass_0_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_icn_pass_1_0/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_icn_pass_1_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_icn_pass_2_0/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_icn_pass_2_0/sim -I../../../../prj.ip_user_files/bd/emu/ip/emu_icn_pass_3_0/src -I../../../../prj.ip_user_files/bd/emu/ip/emu_icn_pass_3_0/sim -I../../../../prj.ip_user_files/bd/emu/sim -I/trinity/shared/pkg/devel/sdaccel/Vivado/2020.2/tps/boost_1_64_0 -I/trinity/shared/pkg/devel/sdaccel/Vivado/2020.2/data/simmodels/xsim/2020.2/lnx64/6.2.0/ext/protobuf/include -I/trinity/shared/pkg/devel/sdaccel/Vivado/2020.2/data/xsim/ip/xtlm/include -I/trinity/shared/pkg/devel/sdaccel/Vivado/2020.2/data/xsim/ip/xtlm_simple_interconnect_v1_0/include -I/trinity/shared/pkg/devel/sdaccel/Vivado/2020.2/data/xsim/ip/common_cpp_v1_0/include -I/trinity/shared/pkg/devel/sdaccel/Vivado/2020.2/data/xsim/ip/emu_perf_common_v1_0/include -work xil_defaultlib -f emu_wrapper_xsc.prj
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [11:06:22] Run run_link: Step vpl: Failed
Time (s): cpu = 00:01:32 ; elapsed = 00:01:56 . Memory (MB): peak = 1670.371 ; gain = 0.000 ; free physical = 338235 ; free virtual = 358341
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
Vitis linkage stage failed
Output /tmp/single_task_vector_add-47f9b65zz08txn/single_task_vector_add-47f9b6.xclbin was not properly produced by previous commands
clang-14: error: sycl-link-vxx command failed with exit code 255 (use -v to see invocation)
clang version 14.0.0 (https://github.com/triSYCL/sycl.git 0d2c6d5fca5e7822c8955955edc31e1acf74d4c0)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /home/stepha27/sycl-workspace/sycl-build/bin
clang-14: note: diagnostic msg: Error generating preprocessed source(s).
```

Any ideas on how to proceed from here? Can this error be debugged further by passing additional flags or something?

### Log file starts here
```
$ clang++ --gcc-toolchain=/trinity/shared/pkg/compiler/gcc/11.2.0 -std=c++20 -fsycl -fsycl-targets=fpga64_hls_hw_emu single_task_vector_add.cpp
Found Vitis version 2020.2
warning: Linking two modules of different target triples: '/trinity/shared/pkg/devel/sdaccel/Vitis_HLS/2020.2/lnx64/lib/libspir64-39-hls.bc' is 'fpga64-xilinx-none' whereas 'llvm-link' is 'fpga64_hls_hw_emu-xilinx-linux'

[... cut ...]
Output /tmp/single_task_vector_add-47f9b65zz08txn/single_task_vector_add-47f9b6.xclbin was not properly produced by previous commands
clang-14: error: sycl-link-vxx command failed with exit code 255 (use -v to see invocation)
clang version 14.0.0 (https://github.com/triSYCL/sycl.git 0d2c6d5fca5e7822c8955955edc31e1acf74d4c0)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /home/stepha27/sycl-workspace/sycl-build/bin
clang-14: note: diagnostic msg: Error generating preprocessed source(s).
```

lforg37, 2022-03-02:
We usually (try to) run the tests only with the most recent available vitis version (that would be 2021.2 as of now). Can you try with it (that requires only 300 GB of disk space 😅) ?

If not, I can try to reproduce the bug and investigate it, but that will probably not be possible before in 3-4 weeks.

j-stephan, 2022-03-02:
Okay, it appears that the Vitis tools are unhappy if I have another GCC compiler toolchain in `PATH` and `LD_LIBRARY_PATH`. So my workflow currently looks like this:

1. Load a modern GCC compiler to enable clang compilation.
2. Enable `SYCL_VXX_KEEP_CLUTTER`.
3. Use clang to compile the test case.
4. Wait until it crashes in the `sycl-link-vxx` step.
5. Unload the modern GCC version.
6. Navigate to the temporary directory mentioned in the log output and rerun the command found in `<tempdir>/07-vxxlink.cmd`.

lforg37, 2022-03-02:
Ok we can try to reproduce these steps with our install to see if this bug still exists in recent vitis version.

j-stephan, 2022-03-02:
Thanks! And I'll see what can be done about a more recent software stack. It also turned out that the node itself is still utilizing XRT 2019.1 which doesn't work well with more recent Vitis versions...

keryell, 2022-03-02:
Try to use the most modern version of everything. :-)

keryell, 2022-03-02:
We do not use the Vitis/Vivado setup because it breaks too many thing, we just use the recipe from https://github.com/triSYCL/sycl/blob/sycl/unified/next/sycl/doc/GettingStartedXilinxFPGA.md#compiling-and-running-a-sycl-application
Are you using a similar setup?

j-stephan, 2022-03-04:
So far I relied on the module system of the cluster to set up everything for me. I have now created a script that bypasses the module system for the Xilinx tools which looks very similar to the environment setup in the link. The results stays the same, though; `vpl` will fail in the `config_hw_emu.compile` step. When diving into the details `/tmp/<build dir>/vxx_link_tmp/link/vivado/vpl/prj/prj.sim/sim_1/behav_waveform/xsim/compile.sh` will fail with the following error message:

```
g++: error trying to exec 'cc1plus': execvp: No such file or directory
```

This error message goes away once I unload the gcc-11 module (which removes the entire gcc toolchain from `PATH` and `LD_LIBRARY_PATH`). Afterwards I can call the command from `/tmp/<build dir>/07-vxxlink.cmd` and it will happily continue (until the node's ancient `xclbinutil` complains about unknown parameters - I hope my IT department will resolve this soon).

j-stephan, 2022-03-10:
IT installed Vitis 2021.2 on the cluster. The issue persists, though - if I load the gcc-11 module the Vitis tools will fail to detect their own `cc1plus`. My current work-around looks like this:

```
$ LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/trinity/shared/pkg/compiler/gcc/11.2.0/lib64 $SYCL_BIN_DIR/clang++ --gcc-toolchain=/trinity/shared/pkg/compiler/gcc/11.2.0 -std=c++20 -fsycl -fsycl-targets=fpga64_hls_hw_emu single_task_vector_add.cpp
```

This lets the HLS flow continue until `xclbinutil` is called.