triSYCL/sycl #175: Unnamed lambda seems not working for `parallel_for`

fxzjshm, 2022-04-28:
**Describe the bug**
Unnamed lambda seems not working for `parallel_for` when compiling. (`single_task` works.)

**To Reproduce**
Use [sycl/test/vitis/simple_tests/answer_42_unnamed.cpp](https://github.com/triSYCL/sycl/blob/sycl/unified/next/sycl/test/vitis/simple_tests/answer_42_unnamed.cpp), but replace
```c++
cgh.single_task([=] { a[0] = 42; });
```
with
```c++
cgh.parallel_for(sycl::range<1>(1), [=] (sycl::id<1> item) { a[item] = 42; });
```
Build command line: (assume the compiler is in `/opt/xilinx-llvm`)
```bash
/opt/xilinx-llvm/bin/clang -fsycl -fsycl-unnamed-lambda -std=c++20 -fsycl-targets=fpga64_hls_hw_emu answer_42_unnamed.cpp -lstdc++ -o answer_42_unnamed-xilinx-hw-emu
```
Log:
<pre>
In file included from answer_42_unnamed.cpp:1:
In file included from /opt/xilinx-llvm/bin/../include/sycl/sycl.hpp:11:
In file included from /opt/xilinx-llvm/bin/../include/sycl/CL/sycl.hpp:16:
In file included from /opt/xilinx-llvm/bin/../include/sycl/CL/sycl/backend.hpp:18:
In file included from /opt/xilinx-llvm/bin/../include/sycl/CL/sycl/detail/backend_traits_opencl.hpp:25:
In file included from /opt/xilinx-llvm/bin/../include/sycl/CL/sycl/queue.hpp:20:
<b>/opt/xilinx-llvm/bin/../include/sycl/CL/sycl/handler.hpp:1231:42: </b><font color="#C01C28"><b>error: </b></font><b>unnamed type &apos;sycl::detail::par_for_to_st&lt;(lambda at answer_42_unnamed.cpp:23:43)&gt;&apos; is invalid; provide a kernel name, or use &apos;-fsycl-unnamed-lambda&apos; to enable unnamed kernel lambdas</b>
    kernel_single_task&lt;ToSingleTaskName&gt;(SingleTaskKernel);
<font color="#26A269"><b>                                         ^</b></font>
<b>/opt/xilinx-llvm/bin/../include/sycl/CL/sycl/handler.hpp:1649:5: </b><font color="#171421"><b>note: </b></font>in instantiation of function template specialization &apos;sycl::handler::parallel_for_lambda_impl&lt;sycl::detail::auto_name, (lambda at answer_42_unnamed.cpp:23:43), 1&gt;&apos; requested here
    parallel_for_lambda_impl&lt;KernelName&gt;(NumWorkItems, std::move(KernelFunc));
<font color="#26A269"><b>    ^</b></font>
<b>answer_42_unnamed.cpp:23:11: </b><font color="#171421"><b>note: </b></font>in instantiation of function template specialization &apos;sycl::handler::parallel_for&lt;sycl::detail::auto_name, (lambda at answer_42_unnamed.cpp:23:43)&gt;&apos; requested here
      cgh.parallel_for(sycl::range&lt;1&gt;(1), [=] (sycl::id&lt;1&gt; item) { a[item] = 42; });
<font color="#26A269"><b>          ^</b></font>
1 error generated. 
</pre>

**Environment (please complete the following information):**

- OS: Ubuntu 20.04 / 22.04
- Target device and vendor: xilinx_u200_gen3x16_xdma_1_202110_1
- DPC++ version: `clang version 15.0.0 (git@github.com:triSYCL/sycl.git 29df7b7d84121137b5ec0f370cf24b2ccfe3683f)` (current sycl/unified/next branch) with Vitis 2021.2
- Dependencies version: XRT build version: 2.14.0

**Additional context**
* On Ubuntu 20.04 the `parallel_for` one doesn't compile; the `single_task` one runs fine
* On Ubuntu 22.04 the `parallel_for` one doesn't compile; the `single_task` one says `No kernel named E_clES2_E10fourty_two_8rZJhPCI was found -46 (CL_INVALID_KERNEL_NAME)` at runtime. 
  Still trying to configure my environment as the latter one seems not related to this issue.

keryell, 2022-04-29:
Actually @Ralender is fixing it right now.

Ralender, 2022-04-29:
All the issues you mentioned should be fixed by https://github.com/triSYCL/sycl/pull/178

Our serialization of `parallel_for`  didnt work for unamed lambdas, the issue you found.
And I think the issue you had with kernels not being found at runtime was caused by a recent change to XRT that broke sycl_vxx_post_link.py because it relies on parsing xclbinutil output.

fxzjshm, 2022-04-30:
Yes, I indeed re-compiled XRT 2.14 myself after upgrade, and I found the abnormal output of `xclbinutil` just now. Downgrading to 2.13 solved this problem.  
Thank you for your timely help!

fxzjshm, 2022-05-03:
Sorry for disturbing you again, the `parallel_for` with `range` works now, but with `nd_range` it still doesn't compile.  
Changing 
```c++
cgh.single_task([=] { a[0] = 42; });
```
to
```c++
cgh.parallel_for(sycl::nd_range<1>(sycl::range<1>(1), sycl::range<1>(1)), [=] (sycl::nd_item<1> item) { a[item.get_global_id(0)] = 42; });
```
Compiler log this time:
<pre>In file included from answer_42_unnamed.cpp:1:
In file included from /opt/xilinx-llvm/bin/../include/sycl/sycl.hpp:11:
In file included from /opt/xilinx-llvm/bin/../include/sycl/CL/sycl.hpp:16:
In file included from /opt/xilinx-llvm/bin/../include/sycl/CL/sycl/backend.hpp:18:
In file included from /opt/xilinx-llvm/bin/../include/sycl/CL/sycl/detail/backend_traits_opencl.hpp:25:
In file included from /opt/xilinx-llvm/bin/../include/sycl/CL/sycl/queue.hpp:20:
<b>/opt/xilinx-llvm/bin/../include/sycl/CL/sycl/handler.hpp:1763:39: </b><font color="#C01C28"><b>error: </b></font><b>no matching function for call to &apos;forward&apos;</b>
                                      std::forward&lt;KernelType&gt;(KernelFunc)}] {
<font color="#26A269"><b>                                      ^~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>answer_42_unnamed.cpp:26:11: </b><font color="#171421"><b>note: </b></font>in instantiation of function template specialization &apos;sycl::handler::parallel_for&lt;sycl::detail::auto_name, (lambda at answer_42_unnamed.cpp:26:81), 1&gt;&apos; requested here
      cgh.parallel_for(sycl::nd_range&lt;1&gt;(sycl::range&lt;1&gt;(1), sycl::range&lt;1&gt;(1)), [=] (sycl::nd_item&lt;1&gt; item) { a[item.get_global_id(0)] = 42; });
<font color="#26A269"><b>          ^</b></font>
<b>/usr/lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/move.h:77:5: </b><font color="#171421"><b>note: </b></font>candidate function template not viable: 1st argument (&apos;const (lambda at answer_42_unnamed.cpp:26:81)&apos;) would lose const qualifier
    forward(typename std::remove_reference&lt;_Tp&gt;::type&amp; __t) noexcept
<font color="#26A269"><b>    ^</b></font>
<b>/usr/lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/move.h:89:5: </b><font color="#171421"><b>note: </b></font>candidate function template not viable: 1st argument (&apos;const (lambda at answer_42_unnamed.cpp:26:81)&apos;) would lose const qualifier
    forward(typename std::remove_reference&lt;_Tp&gt;::type&amp;&amp; __t) noexcept
<font color="#26A269"><b>    ^</b></font>
1 error generated.</pre>

Changing `std::forward<KernelType>(KernelFunc)` to `std::forward<const KernelType>(KernelFunc)` makes it compile temporarily, but I'm not sure if this is proper, nor do I know where does this `const` come from. Could you please look into this compile error?

Ralender, 2022-05-03:
I pushed a fix for this issue in 822e38bb18e0b2b14f26c367b356415b4c17d4ee

fxzjshm, 2022-05-04:
Many thanks for this!

closed 2022-05-04 by fxzjshm