triSYCL/sycl #176: Do not generate DDR bank by default if there is no DDR but HBM

keryell, 2022-04-29:
I have tried the tool-chain on an Alveo U50 (with HBM memory :-) ) according to the documentation with:
```bash
export XILINX_PLATFORM=xilinx_u50_gen3x16_xdma_201920_3
$SYCL_BIN_DIR/clang++ -std=c++20 -fsycl -fsycl-targets=fpga64_hls_hw_emu     single_task_vector_add.cpp -o single_task_vector_add
```
and it fails with
```
rkeryell@rk-xsj:~/Xilinx/Projects/LLVM/worktrees/xilinx/llvm/sycl/test/vitis/simple_tests (doc-2022-04-11)$ $SYCL_BIN_DIR/clang++ -std=c++20 -fsycl -fsycl-targets=fpga64_hls_hw_emu single_task_vector_add.cpp -o single_task_vector_add
Found v++ version 2021.2
Option Map File Used: '/opt/xilinx/Vitis/2021.2/data/vitis/vpp/optMap.xml'

****** v++ v2021.2 (64-bit)
  **** SW Build 3363252 on 2021-10-14-04:41:01
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
	Reports: /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_comp_report/handlerEE_clES2_E3add_eCM5g51B
	Log files: /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_comp_log/handlerEE_clES2_E3add_eCM5g51B
Running Dispatch Server on port: 34087
INFO: [v++ 60-1548] Creating build summary session with primary output /tmp/single_task_vector_add-a16f476x7wb5gt/handlerEE_clES2_E3add_eCM5g51B.xo.compile_summary, at Thu Apr 28 17:22:51 2022
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Thu Apr 28 17:22:51 2022
INFO: [v++ 60-1315] Creating rulecheck session with output '/tmp/single_task_vector_add-a16f476x7wb5gt/vxx_comp_report/handlerEE_clES2_E3add_eCM5g51B/v++_compile_handlerEE_clES2_E3add_eCM5g51B_guidance.html', at Thu Apr 28 17:22:52 2022
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/xilinx_u50_gen3x16_xdma_201920_3.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/hw/hw.xsa'
INFO: [v++ 74-78] Compiler Version string: 2021.2
INFO: [v++ 60-1302] Platform 'xilinx_u50_gen3x16_xdma_201920_3.xpfm' has been explicitly enabled for this release.
INFO: [v++ 60-585] Compiling for hardware emulation target
INFO: [v++ 60-423]   Target device: xilinx_u50_gen3x16_xdma_201920_3
INFO: [v++ 60-242] Creating kernel: 'handlerEE_clES2_E3add_eCM5g51B'

===>The following messages were generated while  performing high-level synthesis for kernel: handlerEE_clES2_E3add_eCM5g51B Log file: /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_comp_tmp/handlerEE_clES2_E3add_eCM5g51B/handlerEE_clES2_E3add_eCM5g51B/vitis_hls.log :
INFO: [v++ 204-61] Pipelining loop 'Loop 1'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 2, loop 'Loop 1'
INFO: [v++ 204-61] Pipelining loop 'Loop 1'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 3, loop 'Loop 1'
INFO: [v++ 200-790] **** Loop Constraint Status: All loop constraints were satisfied.
INFO: [v++ 200-789] **** Estimated Fmax: 411.00 MHz
INFO: [v++ 60-594] Finished kernel compilation
INFO: [v++ 60-244] Generating system estimate report...
INFO: [v++ 60-1092] Generated system estimate report: /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_comp_report/handlerEE_clES2_E3add_eCM5g51B/system_estimate_handlerEE_clES2_E3add_eCM5g51B.xtxt
INFO: [v++ 60-586] Created /tmp/single_task_vector_add-a16f476x7wb5gt/handlerEE_clES2_E3add_eCM5g51B.xo
INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command. 
    vitis_analyzer /tmp/single_task_vector_add-a16f476x7wb5gt/handlerEE_clES2_E3add_eCM5g51B.xo.compile_summary 
INFO: [v++ 60-791] Total elapsed time: 0h 0m 48s
INFO: [v++ 60-1653] Closing dispatch client.
Option Map File Used: '/opt/xilinx/Vitis/2021.2/data/vitis/vpp/optMap.xml'

****** v++ v2021.2 (64-bit)
  **** SW Build 3363252 on 2021-10-14-04:41:01
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ link can be found at:
	Reports: /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_link_report/link
	Log files: /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_link_log/link
Running Dispatch Server on port: 33135
INFO: [v++ 60-1548] Creating build summary session with primary output /tmp/single_task_vector_add-a16f476x7wb5gt/single_task_vector_add-a16f47.xclbin.link_summary, at Thu Apr 28 17:23:41 2022
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Thu Apr 28 17:23:41 2022
INFO: [v++ 60-1315] Creating rulecheck session with output '/tmp/single_task_vector_add-a16f476x7wb5gt/vxx_link_report/link/v++_link_single_task_vector_add-a16f47_guidance.html', at Thu Apr 28 17:23:42 2022
INFO: [v++ 60-895]   Target platform: /opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/xilinx_u50_gen3x16_xdma_201920_3.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/hw/hw.xsa'
INFO: [v++ 74-78] Compiler Version string: 2021.2
INFO: [v++ 60-1302] Platform 'xilinx_u50_gen3x16_xdma_201920_3.xpfm' has been explicitly enabled for this release.
INFO: [v++ 60-629] Linking for hardware emulation target
INFO: [v++ 60-423]   Target device: xilinx_u50_gen3x16_xdma_201920_3
INFO: [v++ 60-1332] Run 'run_link' status: Not started
INFO: [v++ 60-1443] [17:23:45] Run run_link: Step system_link: Started
INFO: [v++ 60-1453] Command Line: system_link --xo /tmp/single_task_vector_add-a16f476x7wb5gt/handlerEE_clES2_E3add_eCM5g51B.xo -keep --config /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_link_tmp/link/int/syslinkConfig.ini --xpfm /opt/xilinx/platforms/xilinx_u50_gen3x16_xdma_201920_3/xilinx_u50_gen3x16_xdma_201920_3.xpfm --target emu --output_dir /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_link_tmp/link/int --temp_dir /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_link_tmp/link/sys_link
INFO: [v++ 60-1454] Run Directory: /tmp/single_task_vector_add-a16f476x7wb5gt/vxx_link_tmp/link/run_link
INFO: [SYSTEM_LINK 60-1316] Initiating connection to rulecheck server, at Thu Apr 28 17:23:46 2022
INFO: [SYSTEM_LINK 82-70] Extracting xo v3 file /tmp/single_task_vector_add-a16f476x7wb5gt/handlerEE_clES2_E3add_eCM5g51B.xo
[... 27 lines cut ...]
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
Vitis linkage stage failed
Traceback (most recent call last):
  File "/home/rkeryell/Xilinx/Projects/LLVM/worktrees/xilinx/llvm/build/bin/sycl_vxx.py", line 660, in <module>
    if (not main()):
  File "/home/rkeryell/Xilinx/Projects/LLVM/worktrees/xilinx/llvm/build/bin/sycl_vxx.py", line 654, in main
    return cd.drive_compilation()
  File "/home/rkeryell/Xilinx/Projects/LLVM/worktrees/xilinx/llvm/build/bin/sycl_vxx.py", line 403, in drive_compilation
    shutil.copy2(final, self.outpath)
  File "/usr/lib/python3.10/shutil.py", line 434, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.10/shutil.py", line 233, in copyfile
    if _samefile(src, dst):
  File "/usr/lib/python3.10/shutil.py", line 210, in _samefile
    return os.path.samefile(src, dst)
  File "/usr/lib/python3.10/genericpath.py", line 100, in samefile
    s1 = os.stat(f1)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType
clang-15: error: sycl-link-vxx command failed with exit code 1 (use -v to see invocation)
```
because our internal pass just guess that there is only `DDR[0]` bank by default instead of HBM.
Also there is no HBM decoration in `sycl/include/sycl/ext/xilinx/fpga/memory_properties.hpp`.

keryell, 2022-04-29:
@lforg37 I have installed the `xilinx_u50_gen3x16_xdma_201920_3` platform on `xsjsycl41` so it can be tried in `hw_emu`.

rgioiosa78, 2022-05-09:
Hello everyone! I'm having the same issue on a Versal board, is there any update on this? Thanks for your great work!

Ralender, 2022-05-09:
> Hello everyone! I'm having the same issue on a Versal board, is there any update on this? Thanks for your great work!

There is a PR(#177) trying to fix this issue  but it has not yet been tested due to configuration issues with our testing machine.

rgioiosa78, 2022-05-10:
Thanks for the quick answer! I'll be happy to test on my system once this is ready to go.

keryell, 2022-05-10:
@lforg37 got the U50 card working on our machine so we expect https://github.com/triSYCL/sycl/pull/177 to make some progress.