triSYCL/sycl #181: Implement USM support for XRT plugin

keryell, 2022-05-13:


keryell, 2023-06-15:
No, but we are thinking to it.
I think just the `sycl::malloc_device` could be done, perhaps `sycl::malloc_host` too. Do not even think about `sycl::malloc_shared`.
What model do you need?
@gauthier The idea would be to use a `xrt::bo` and just call `address()` member function in the XRT PI to implement `sycl::malloc_device`.
There are the flags `bo::flags::device_only`, `host_only` and `svm`.
@stsoe I hope that `host_only` still allows the device to use the buffer through PCIe in an Alveo U200 for example?
Probably they might need some specific attribute on the Vitis HLS kernel arguments too.

j-stephan, 2023-06-16:
Thanks for letting me know. For the beginning, `malloc_{device,host}` would be sufficient.