triSYCL/sycl #202: /unified/next stopped compiling with latest updates

gogo2, 2022-11-08:
#### Bug description
After latest changes in `https://github.com/triSYCL/sycl/blob/sycl/unified/next/sycl/plugins/xrt/pi_xrt.cpp` the toolchain stopped compiling with the following:
```
error: use of deleted function ‘_pi_mem::~_pi_mem()’
note: ‘_pi_mem::~_pi_mem()’ is implicitly deleted because the default definition would be ill-formed:
error: union member ‘no_destroy<xrt::bo>::<unnamed union>::data’ with non-trivial ‘xrt::bo::~bo()’ 
```
and similar errors for other classes and destructors (full output below)

#### Environment
* Ubuntu 22.04
* xrt 2.15 (latest master)
* gcc-11, gcc-12, clang-15

#### Messages

```
AILED: tools/sycl/plugins/xrt/CMakeFiles/pi_xrt.dir/pi_xrt.cpp.o                                                                                                                                                                              
In file included from /hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:66:                                                                                                                                                     
/opt/xilinx/xrt/include/xrt/xrt_kernel.h:117:5: warning: ‘virtual const char* xrt::run::command_error::what() const’ can be marked override [-Wsuggest-override]                                                                               
  117 |     what() const noexcept;                                                                                                                                                                                                             
      |     ^~~~                                                                                                                                                                                                                               
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp: In constructor ‘_pi_mem::_pi_mem(_pi_context*, _pi_mem::_mem)’:                                                                                                              
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:447:59: error: use of deleted function ‘no_destroy<xrt::bo>::~no_destroy()’                                                                                                   
  447 |   _pi_mem(_pi_context *ctx, _mem m) : context_(ctx), mem(m) {}                                                                                                                                                                         
      |                                                           ^                                                                                                                                                                            
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:346:29: note: ‘no_destroy<xrt::bo>::~no_destroy()’ is implicitly deleted because the default definition would be ill-formed:
  346 | template <typename T> class no_destroy {
      |                             ^~~~~~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:348:7: error: union member ‘no_destroy<xrt::bo>::<unnamed union>::data’ with non-trivial ‘xrt::bo::~bo()’
  348 |     T data;
      |       ^~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp: In constructor ‘_pi_kernel::_pi_kernel(ref_counted_ref<_pi_program>, _pi_kernel::native_type, xrt::xclbin::kernel)’:
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:560:21: error: use of deleted function ‘no_destroy<xrt::kernel>::~no_destroy()’
  560 |       : prog_(ctx), kernel_(std::move(kern)),
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:346:29: note: ‘no_destroy<xrt::kernel>::~no_destroy()’ is implicitly deleted because the default definition would be ill-formed:
  346 | template <typename T> class no_destroy {
      |                             ^~~~~~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:348:7: error: union member ‘no_destroy<xrt::kernel>::<unnamed union>::data’ with non-trivial ‘xrt::kernel::~kernel()’
  348 |     T data;
      |       ^~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp: In function ‘pi_result piPluginInit(pi_plugin*)’:
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:2180:42: warning: cast between incompatible function types from ‘_pi_result (*)(long unsigned int, _pi_context*, _pi_queue**, bool)’ to ‘pi_result (*)(pi_native_handle, pi_context, pi_device, bool, _pi_queue**)’ {aka ‘_pi_result (*)(long unsigned int, _pi_context*, _pi_device*, bool, _pi_queue**)’} [-Wcast-function-type]
 2180 |   (PluginInit->PiFunctionTable).pi_api = (decltype(                            \
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 2181 |       &::pi_api))xrt_pi_call_wrapper<decltype(&xrt_api), &xrt_api>::call;
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:2216:3: note: in expansion of macro ‘_PI_CL’
 2216 |   _PI_CL(piextQueueCreateWithNativeHandle, xrt_piextQueueCreateWithNativeHandle)
      |   ^~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:2180:42: warning: cast between incompatible function types from ‘_pi_result (*)(long unsigned int, _pi_mem**)’ to ‘pi_result (*)(pi_native_handle, pi_context, bool, _pi_mem**)’ {aka ‘_pi_result (*)(long unsigned int, _pi_context*, bool, _pi_mem**)’} [-Wcast-function-type]
 2180 |   (PluginInit->PiFunctionTable).pi_api = (decltype(                            \
      |                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 2181 |       &::pi_api))xrt_pi_call_wrapper<decltype(&xrt_api), &xrt_api>::call;
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:2226:3: note: in expansion of macro ‘_PI_CL’
 2226 |   _PI_CL(piextMemCreateWithNativeHandle, xrt_piextMemCreateWithNativeHandle)
      |   ^~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp: In instantiation of ‘uint32_t ref_counted_base<base>::decrement_reference_count() [with base = _pi_mem; uint32_t = unsigned int]’:
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:128:35:   required from ‘std::void_t<decltype (declval<T>().decrement_reference_count())> decr_ref_count(T*) [with T = _pi_mem; std::void_t<decltype (declval<T>().decrement_reference_count())> = void; decltype (declval<T>().decrement_reference_count()) = unsigned int]’
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:1302:17:   required from here
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:107:7: error: use of deleted function ‘_pi_mem::~_pi_mem()’
  107 |       delete static_cast<base *>(this);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:424:8: note: ‘_pi_mem::~_pi_mem()’ is implicitly deleted because the default definition would be ill-formed:
  424 | struct _pi_mem : ref_counted_base<_pi_mem> {
      |        ^~~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:424:8: error: use of deleted function ‘no_destroy<xrt::bo>::~no_destroy()’
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp: In instantiation of ‘uint32_t ref_counted_base<base>::decrement_reference_count() [with base = _pi_kernel; uint32_t = unsigned int]’:
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:128:35:   required from ‘std::void_t<decltype (declval<T>().decrement_reference_count())> decr_ref_count(T*) [with T = _pi_kernel; std::void_t<decltype (declval<T>().decrement_reference_count())> = void; decltype (declval<T>().decrement_reference_count()) = unsigned int]’
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:1794:17:   required from here
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:107:7: error: use of deleted function ‘_pi_kernel::~_pi_kernel()’
  107 |       delete static_cast<base *>(this);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:547:8: note: ‘_pi_kernel::~_pi_kernel()’ is implicitly deleted because the default definition would be ill-formed:
  547 | struct _pi_kernel : ref_counted_base<_pi_kernel> {
      |        ^~~~~~~~~~
/hac/home/bsobol/install_git/triSYCL/sycl/plugins/xrt/pi_xrt.cpp:547:8: error: use of deleted function ‘no_destroy<xrt::kernel>::~no_destroy()’

```

Ralender, 2022-11-14:
I found it this issue while making https://github.com/triSYCL/sycl/pull/205. so this should be fixed with this PR.

gogo2, 2022-11-15:
> I found it this issue while making #205. so this should be fixed with this PR.
Your branch compiles, thanks!

closed 2022-11-15 by gogo2