triSYCL/sycl #206: Aggregate initialziation crashes opt

gogo2, 2022-11-29:
### Problem description
When I try to compile such code:
```cpp
struct Line {
    float slope;
    float intercept;
};

int main() {
    sycl::queue queue{};
    {
        queue.submit([&](sycl::handler &cgh) {
            cgh.single_task([]() {
                Line l1{0, 0};
            });
        });
    }
}
```
it fails with a opt crash in sycl->HLS conversion step ("Error in sycl->HLS conversion").

It works if I _disable_ aggregate initialization by defining construstor `Line(float s, float i) : slope{s}, intercept{i} {}` or using default member initializer `float intercept = 0`.

But the interesting thing is that it also works if pass something other than zeros to aggregate:
```cpp 
Line l1{0, 1};
```
I believe this also affect `std::array`.

Any chance for fixing this issue or at least some hint on where to look?

### Setup

- OS: Ubuntu 22.04
- triSYCL/sycl: latest unified/next

Ralender, 2022-12-01:
I pushed a commit(7a775748caadf44261a6a1e1b2b74f45afe893e5) that should fix this issue.

LLVM loves making anything it can into a memset but memset are handled poorly by the HLS backend.
so our passes will turn memset back stores of constants. this was a bug into the code generating a constant that need to be stored.
it doesn't happened for {0, 1} because it cannot be turned into a memset.

gogo2, 2022-12-06:
This fix does the job for this issue, many thanks for help!

closed 2022-12-06 by gogo2