triSYCL/sycl #212: fPIC error when compiling as shared library

gogo2, 2022-12-09:
### Some background

First of all, sorry for creating so many issues recently. I'm working on a paper about the project I mentioned some time ago in https://github.com/triSYCL/sycl/issues/123 (now fully finished with about 1000 lines of accelerated code and 7 kernels implemented as callable objects) and having the algorithm compiled and working on FPGA would be a very nice addition.
I hope that my reports give a comprehensive description of the issues I have and they somehow contribute to overall development and improvement of your SYCL implementation.

### Problem description

This time, when I try to compile sycl-accelerated part as a shared library (and consequently with -fPIC flag) , during the `_asm_ir` step I get a llvm-as error with message: 
`assembly parsed, but does not verify as correct!
invalid behavior operand in module flag (unexpected constant)
i32 8`

I have found out that that the direct cause is the line containing `!{i32 8, !"PIC Level", i32 2}`  in `*_kernels_linked.opt.ll` file.
Changing the above `!{i32 8, !"PIC Level", i32 2}` helps, library compiles successfully and hw_emu gives correct results, but i'm not an llvm expert by any means and I have no idea what is the correct value and maybe it is just luck that it worked so far.

### Setup

- OS: Ubuntu 22.04
- Vitis 2022.2
- xrt 202220.2.14.368
- triSYCL/sycl: latest unified/next with recent lowering fix
- HLS flow / hw-emu mode / target platform: latest u280-whatever-it-is-called

Ralender, 2022-12-09:
> First of all, sorry for creating so many issues recently. I'm working on a paper about the project I mentioned some time ago in https://github.com/triSYCL/sycl/issues/123 (now fully finished with [...]
I hope that my reports give a comprehensive description of the issues I have and they somehow contribute to overall development and improvement of your SYCL implementation.

no worries I know this tool is far from perfect and you are helping making it better.
I will look into this issue.

Ralender, 2022-12-14:
this https://github.com/triSYCL/sycl/pull/213 should fix your issue.

since compiling in PIC is useless for device code. I edited the driver to not compile device code in PIC mode.

gogo2, 2022-12-15:
> this #213 should fix your issue.
> 
> since compiling in PIC is useless for device code. I edited the driver to not compile device code in PIC mode.

I confirm. Thanks again!

closed 2022-12-15 by gogo2