triSYCL/sycl #219: kernel_param doesn't work with parallel_for

gogo2, 2023-01-01:
## Describe the bug 
As in description, compilation of kernel_param when used with paralle_for wails with error:
```
/home/bsobol/work/trisycl/build/bin/../include/sycl/types.hpp:2445:17: error: ambiguous partial specializations of 'is_device_copyable<sycl::detail::assume_device_copyable_wrapper<sycl::detail::KernelDecorator<(lambda at /home/bsobol/work/trisycl_tests/tests/test_11_kernel_param.cpp:16:63), void ((lambda at /home/bsobol/work/trisycl_tests/tests/test_11_kernel_param.cpp:16:63)::*)(sycl::id<1>) const, sycl::detail::cstr<char, 'k', 'e', 'r', 'n', 'e', 'l', '_', 'p', 'a', 'r', 'a', 'm'>, sycl::detail::StrPacker<sycl::detail::cstr<char, '-', '-', 'k', 'e', 'r', 'n', 'e', 'l', '_', 'f', 'r', 'e', 'q', 'u', 'e', 'n', 'c', 'y', ' ', '2', '0', '0'>>::Str>>>'
  static_assert(is_device_copyable<FieldT>::value ||
                ^
```

Here is a simple code sample for reproducing the issue:
```
#include <CL/sycl.hpp>
#include <sycl/ext/xilinx/fpga.hpp>

using namespace cl;
using namespace sycl::ext::xilinx;

int main() {
    sycl::queue queue{};
    sycl::range<1> rng{4};
    {
        queue.submit([&](sycl::handler &cgh) {
            cgh.single_task<class Kernel1>(kernel_param([=]() {}, "--kernel_frequency 200"_cstr));
        });

        queue.submit([&](sycl::handler &cgh) {
            cgh.parallel_for<class Kernel2>(rng, kernel_param([=](sycl::id<1> id) {}, "--kernel_frequency 200"_cstr));
        });
    }

}
```

## Environment 
triSYCL/sycl: latest unified/next at https://github.com/triSYCL/sycl/commit/65e2ec51b590e5aa78f0f9f413ba78b2ae8e70d6
SYCL code compiled with C++20 flag enabled

Ralender, 2023-01-04:
I have a fix on https://github.com/triSYCL/sycl/pull/216

Ralender, 2023-01-05:
Ths fix is now in https://github.com/triSYCL/sycl/pull/220

closed 2023-01-09 by gogo2