triSYCL/sycl #236: Compilation issue with aie++

keryell, 2023-08-17:
With latest `unified/next` I get this error:
```
python3 ./buildbot/configure.py \
 -o build-release \
 --cuda --hip --xrt \
 --shared-libs \
 --use-lld \
 --cmake-opt="-DCMAKE_EXPORT_COMPILE_COMMANDS=1" \
 -t Release \
 --cmake-gen Ninja \
 --llvm-external-projects=mlir,clang-tools-extra \
 --cmake-opt="-DCMAKE_C_COMPILER=/usr/lib/llvm-18/bin/clang" \
 --cmake-opt="-DCMAKE_CXX_COMPILER=/usr/lib/llvm-18/bin/clang++" \
 --cmake-opt="-DCMAKE_PREFIX_PATH=/storage/gauthier/flopoco/install" \
 --cmake-opt="-DLLVM_TARGETS_TO_BUILD=X86;AArch64;ARM;NVPTX;AMDGPU"

python3 ./buildbot/configure.py \
 -o build-release \
 --cuda --hip --xrt \
 --shared-libs \
 --use-lld \
 --cmake-opt="-DCMAKE_EXPORT_COMPILE_COMMANDS=1" \
 -t Release \
 --cmake-gen Ninja \
 --llvm-external-projects=mlir,clang-tools-extra \
 --cmake-opt="-DCMAKE_C_COMPILER=/usr/lib/llvm-18/bin/clang" \
 --cmake-opt="-DCMAKE_CXX_COMPILER=/usr/lib/llvm-18/bin/clang++" \
 --cmake-opt="-DCMAKE_PREFIX_PATH=/storage/gauthier/flopoco/install" \
 --cmake-opt="-DLLVM_TARGETS_TO_BUILD=X86;AArch64;ARM;NVPTX;AMDGPU"

cmake --build build-release --verbose --parallel `nproc`
# Relaunch with 1 thread to figure-out the failure
rkeryell@xsjsycl41:/var/tmp/rkeryell/SYCL/llvm (sycl/unified/next)$ cmake --build build-release --verbose --parallel 
1                                                                                                                    
[0/2] /usr/bin/cmake -P /var/tmp/rkeryell/SYCL/llvm/build-release/CMakeFiles/VerifyGlobs.cmake                    
[1/5806] : && /usr/lib/llvm-18/bin/clang++ -fPIC -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werro
r=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmiss
ing-field-initializers -pedantic -Wno-long-long -Wc++98-compat-extra-semi -Wimplicit-fallthrough -Wcovered-switch-def
ault -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmislead
ing-indentation -Wctad-maybe-unsupported -fdiagnostics-color -ffunction-sections -fdata-sections -O3 -DNDEBUG  -Wl,-z
,defs -Wl,-z,nodelete -fuse-ld=lld -Wl,--color-diagnostics   -Wl,--gc-sections  -Wl,--version-script,"/var/tmp/rkerye
ll/SYCL/llvm/build-release/tools/lto/LTO.exports" -shared -Wl,-soname,libLTO.so.16git -o lib/libLTO.so.16git tools/lt
o/CMakeFiles/LTO.dir/LTODisassembler.cpp.o tools/lto/CMakeFiles/LTO.dir/lto.cpp.o  -Wl,-rpath,"\$ORIGIN/../lib"  lib/
libLLVMX86AsmParser.so.16git  lib/libLLVMAArch64AsmParser.so.16git  lib/libLLVMARMAsmParser.so.16git  lib/libLLVMAMDG
PUAsmParser.so.16git  lib/libLLVMX86CodeGen.so.16git  lib/libLLVMAArch64CodeGen.so.16git  lib/libLLVMARMCodeGen.so.16
git  lib/libLLVMNVPTXCodeGen.so.16git  lib/libLLVMAMDGPUCodeGen.so.16git  lib/libLLVMX86Desc.so.16git  lib/libLLVMNVP
TXDesc.so.16git  lib/libLLVMX86Disassembler.so.16git  lib/libLLVMAArch64Disassembler.so.16git  lib/libLLVMARMDisassem
bler.so.16git  lib/libLLVMAMDGPUDisassembler.so.16git  lib/libLLVMX86Info.so.16git  lib/libLLVMNVPTXInfo.so.16git  li
b/libLLVMLTO.so.16git  lib/libLLVMAArch64Desc.so.16git  lib/libLLVMAArch64Info.so.16git  lib/libLLVMARMDesc.so.16git 
 lib/libLLVMARMInfo.so.16git  lib/libLLVMAMDGPUDesc.so.16git  lib/libLLVMAMDGPUInfo.so.16git  lib/libLLVMMCDisassembl
er.so.16git  lib/libLLVMCodeGen.so.16git  lib/libLLVMTarget.so.16git  lib/libLLVMMC.so.16git  lib/libLLVMBitReader.so
.16git  lib/libLLVMCore.so.16git  lib/libLLVMSupport.so.16git  -Wl,-rpath-link,/var/tmp/rkeryell/SYCL/llvm/build-rele
ase/lib && :
FAILED: lib/libLTO.so.16git 
: && /usr/lib/llvm-18/bin/clang++ -fPIC -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-ti
me -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field
-initializers -pedantic -Wno-long-long -Wc++98-compat-extra-semi -Wimplicit-fallthrough -Wcovered-switch-default -Wno
-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-inden
tation -Wctad-maybe-unsupported -fdiagnostics-color -ffunction-sections -fdata-sections -O3 -DNDEBUG  -Wl,-z,defs -Wl
,-z,nodelete -fuse-ld=lld -Wl,--color-diagnostics   -Wl,--gc-sections  -Wl,--version-script,"/var/tmp/rkeryell/SYCL/l
lvm/build-release/tools/lto/LTO.exports" -shared -Wl,-soname,libLTO.so.16git -o lib/libLTO.so.16git tools/lto/CMakeFi
les/LTO.dir/LTODisassembler.cpp.o tools/lto/CMakeFiles/LTO.dir/lto.cpp.o  -Wl,-rpath,"\$ORIGIN/../lib"  lib/libLLVMX8
6AsmParser.so.16git  lib/libLLVMAArch64AsmParser.so.16git  lib/libLLVMARMAsmParser.so.16git  lib/libLLVMAMDGPUAsmPars
er.so.16git  lib/libLLVMX86CodeGen.so.16git  lib/libLLVMAArch64CodeGen.so.16git  lib/libLLVMARMCodeGen.so.16git  lib/
libLLVMNVPTXCodeGen.so.16git  lib/libLLVMAMDGPUCodeGen.so.16git  lib/libLLVMX86Desc.so.16git  lib/libLLVMNVPTXDesc.so
.16git  lib/libLLVMX86Disassembler.so.16git  lib/libLLVMAArch64Disassembler.so.16git  lib/libLLVMARMDisassembler.so.1
6git  lib/libLLVMAMDGPUDisassembler.so.16git  lib/libLLVMX86Info.so.16git  lib/libLLVMNVPTXInfo.so.16git  lib/libLLVM
LTO.so.16git  lib/libLLVMAArch64Desc.so.16git  lib/libLLVMAArch64Info.so.16git  lib/libLLVMARMDesc.so.16git  lib/libL
LVMARMInfo.so.16git  lib/libLLVMAMDGPUDesc.so.16git  lib/libLLVMAMDGPUInfo.so.16git  lib/libLLVMMCDisassembler.so.16g
it  lib/libLLVMCodeGen.so.16git  lib/libLLVMTarget.so.16git  lib/libLLVMMC.so.16git  lib/libLLVMBitReader.so.16git  l
ib/libLLVMCore.so.16git  lib/libLLVMSupport.so.16git  -Wl,-rpath-link,/var/tmp/rkeryell/SYCL/llvm/build-release/lib &
& :
ld.lld: error: version script assignment of 'LLVM_16' to symbol 'LLVMCreateDisasm' failed: symbol not defined
ld.lld: error: version script assignment of 'LLVM_16' to symbol 'LLVMCreateDisasmCPU' failed: symbol not defined
ld.lld: error: version script assignment of 'LLVM_16' to symbol 'LLVMDisasmDispose' failed: symbol not defined
ld.lld: error: version script assignment of 'LLVM_16' to symbol 'LLVMDisasmInstruction' failed: symbol not defined
ld.lld: error: version script assignment of 'LLVM_16' to symbol 'LLVMSetDisasmOptions' failed: symbol not defined
ld.lld: error: version script assignment of 'LLVM_16' to symbol 'LLVMCreateDisasmCPUFeatures' failed: symbol not defined
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

Curious this `LLVM_16`.

Ralender, 2023-08-17:
I just tried the latest unified next and have no issue compilling with:
```bash
python3 ./buildbot/configure.py \
 -o build-release \
 --shared-libs \
 --use-lld \
 --cmake-opt="-DCMAKE_EXPORT_COMPILE_COMMANDS=1" \
 -t Release \
 --cmake-gen Ninja \
 --llvm-external-projects=mlir,clang-tools-extra \
 --cmake-opt="-DCMAKE_C_COMPILER=/usr/bin/clang-14" \
 --cmake-opt="-DCMAKE_CXX_COMPILER=/usr/bin/clang++-14" \
 --cmake-opt="-DLLVM_TARGETS_TO_BUILD=X86;AArch64;ARM;NVPTX;AMDGPU"
```

note I removed `--cuda --hip --xrt` because I dont have the necessary runtimes installed. and I am using clang 14

keryell, 2023-08-17:
@Ralender mentioned https://github.com/llvm/llvm-project/issues/60568

keryell, 2023-08-17:
I removed `lld` and almost everything and it compiled with:
```
python3 ./buildbot/configure.py  -o build-release  --xrt  --shared-libs  --cmake-opt="-DCMAKE_EXPORT_COMPILE_COMMANDS=1"  -t Release  --cmake-gen Ninja  --cmake-opt="-DCMAKE_C_COMPILER=/usr/lib/llvm-18/bin/clang"  --cmake-opt="-DCMAKE_CXX_COMPILER=/usr/lib/llvm-18/bin/clang++" --cmake-opt="-DLLVM_TARGETS_TO_BUILD=X86;AArch64;ARM"
```
Otherwise there was a failure otherwise with libstd++12 on `clang-tools-extra`:
```
python3 ./buildbot/configure.py  -o build-release  --xrt  --shared-libs  --cmake-opt="-DCMAKE_EXPORT_COMPILE_COMMANDS=1"  -t Release  --cmake-gen Ninja  --llvm-external-projects=clang-tools-extra  --cmake-opt="-DCMAKE_C_COMPILER=/usr/lib/llvm-18/bin/clang"  --cmake-opt="-DCMAKE_CXX_COMPILER=/usr/lib/llvm-18/bin/clang++" --cmake-opt="-DLLVM_TARGETS_TO_BUILD=X86;AArch64;ARM"
cmake --build build-release --verbose --parallel `nproc`
[0/2] /usr/bin/cmake -P /var/tmp/rkeryell/SYCL/llvm/build-release/CMakeFiles/VerifyGlobs.cmake
[1/1046] /usr/lib/llvm-18/bin/clang++ -DGTEST_HAS_RTTI=0 -D_DEBUG -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/var/tmp/rkeryell/SYCL/llvm/build-release/tools/clang/tools/extra/clang-tidy/plugin -I/var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin -I/var/tmp/rkeryell/SYCL/llvm/build-release/tools/clang/tools/extra/clang-tidy -I/var/tmp/rkeryell/SYCL/llvm/llvm/../llvm-spirv/include -I/var/tmp/rkeryell/SYCL/llvm/clang/include -I/var/tmp/rkeryell/SYCL/llvm/build-release/tools/clang/include -I/var/tmp/rkeryell/SYCL/llvm/build-release/include -I/var/tmp/rkeryell/SYCL/llvm/llvm/include -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wc++98-compat-extra-semi -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -Wno-nested-anon-types -O3 -DNDEBUG  -fno-exceptions -fno-rtti -UNDEBUG -std=c++17 -MD -MT tools/clang/tools/extra/clang-tidy/plugin/CMakeFiles/obj.clangTidyPlugin.dir/ClangTidyPlugin.cpp.o -MF tools/clang/tools/extra/clang-tidy/plugin/CMakeFiles/obj.clangTidyPlugin.dir/ClangTidyPlugin.cpp.o.d -o tools/clang/tools/extra/clang-tidy/plugin/CMakeFiles/obj.clangTidyPlugin.dir/ClangTidyPlugin.cpp.o -c /var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin/ClangTidyPlugin.cpp
FAILED: tools/clang/tools/extra/clang-tidy/plugin/CMakeFiles/obj.clangTidyPlugin.dir/ClangTidyPlugin.cpp.o 
/usr/lib/llvm-18/bin/clang++ -DGTEST_HAS_RTTI=0 -D_DEBUG -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/var/tmp/rkeryell/SYCL/llvm/build-release/tools/clang/tools/extra/clang-tidy/plugin -I/var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin -I/var/tmp/rkeryell/SYCL/llvm/build-release/tools/clang/tools/extra/clang-tidy -I/var/tmp/rkeryell/SYCL/llvm/llvm/../llvm-spirv/include -I/var/tmp/rkeryell/SYCL/llvm/clang/include -I/var/tmp/rkeryell/SYCL/llvm/build-release/tools/clang/include -I/var/tmp/rkeryell/SYCL/llvm/build-release/include -I/var/tmp/rkeryell/SYCL/llvm/llvm/include -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wc++98-compat-extra-semi -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wsuggest-override -Wstring-conversion -Wmisleading-indentation -Wctad-maybe-unsupported -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -Wno-nested-anon-types -O3 -DNDEBUG  -fno-exceptions -fno-rtti -UNDEBUG -std=c++17 -MD -MT tools/clang/tools/extra/clang-tidy/plugin/CMakeFiles/obj.clangTidyPlugin.dir/ClangTidyPlugin.cpp.o -MF tools/clang/tools/extra/clang-tidy/plugin/CMakeFiles/obj.clangTidyPlugin.dir/ClangTidyPlugin.cpp.o.d -o tools/clang/tools/extra/clang-tidy/plugin/CMakeFiles/obj.clangTidyPlugin.dir/ClangTidyPlugin.cpp.o -c /var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin/ClangTidyPlugin.cpp
In file included from /var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin/ClangTidyPlugin.cpp:9:
In file included from /var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin/../ClangTidy.h:12:
In file included from /var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin/../ClangTidyDiagnosticConsumer.h:12:
In file included from /var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin/../ClangTidyOptions.h:12:
In file included from /var/tmp/rkeryell/SYCL/llvm/llvm/include/llvm/ADT/IntrusiveRefCntPtr.h:66:
In file included from /usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/memory:75:
/usr/lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/bits/unique_ptr.h:1065:34: error: allocating an object of abstract class type 'clang::tidy::ClangTidyPluginAction'
 1065 |     { return unique_ptr<_Tp>(new _Tp(std::forward<_Args>(__args)...)); }
      |                                  ^
[... cut ...]
/var/tmp/rkeryell/SYCL/llvm/clang-tools-extra/clang-tidy/plugin/ClangTidyPlugin.cpp:84:5: note: in instantiation of member function 'llvm::Registry<clang::PluginASTAction>::Add<clang::tidy::ClangTidyPluginAction>::Add' requested here
   84 |     X("clang-tidy", "clang-tidy");
      |     ^
/var/tmp/rkeryell/SYCL/llvm/clang/include/clang/Frontend/FrontendAction.h:290:16: note: unimplemented pure virtual method 'takeMainActionToReplace' in 'ClangTidyPluginAction'
  290 |   virtual void takeMainActionToReplace(std::unique_ptr<FrontendAction> Old) = 0;
      |                ^
1 error generated.
ninja: build stopped: subcommand failed.
```